eelcocramer/node-bluetooth-serial-port #54: npm install gyp build error: call to 'New' is ambiguous

davydka, 2015-01-22:
When trying to run npm install bluetooth-serial-port I get the following error on osx:

CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/BluetoothWorker.o
  CC(target) Release/obj.target/BluetoothSerialPort/src/osx/pipe.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/BluetoothDeviceResources.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/BluetoothSerialPort.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/BTSerialPortBinding.o
In file included from ../src/osx/BTSerialPortBinding.mm:14:
In file included from ../node_modules/nan/nan.h:63:
../node_modules/nan/nan_new.h:184:10: error: call to 'New' is ambiguous
  return NanIntern::Factory<T>::New(arg0);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
../src/osx/BTSerialPortBinding.mm:124:19: note: in instantiation of function template specialization 'NanNew<v8::Integer, unsigned long>' requested here
        argv[1] = NanNew<v8::Integer>(data->result);
                  ^
../node_modules/nan/nan_new.h:86:26: note: candidate function
  static inline return_t New(int32_t value);
                         ^
../node_modules/nan/nan_new.h:87:26: note: candidate function
  static inline return_t New(uint32_t value);
                         ^
1 error generated.
make: *** [Release/obj.target/BluetoothSerialPort/src/osx/BTSerialPortBinding.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Darwin 13.4.0
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "build"
gyp ERR! cwd /Users/davidwhitely/Sites/comcast-xr16/bluetooth-serial/node_modules/bluetooth-serial-port
gyp ERR! node -v v0.10.35
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok 

npm ERR! bluetooth-serial-port@1.2.0 install: `node-gyp configure build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bluetooth-serial-port@1.2.0 install script.
npm ERR! This is most likely a problem with the bluetooth-serial-port package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp configure build
npm ERR! You can get their info via:
npm ERR!     npm owner ls bluetooth-serial-port
npm ERR! There is likely additional logging output above.
npm ERR! System Darwin 13.4.0
npm ERR! command "node" "/usr/local/bin/npm" "install" "bluetooth-serial-port"
npm ERR! cwd /Users/davidwhitely/Sites/comcast-xr16/bluetooth-serial
npm ERR! node -v v0.10.35
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0

eelcocramer, 2015-01-22:
Thanks for bringing this to my attention. Is a similar issue as https://github.com/rvagg/nan/issues/233. I will issue a fix today.

eelcocramer, 2015-01-22:
Just released 1.2.1 that fixes this issue.

closed 2015-01-22 by eelcocramer

davydka, 2015-01-22:
Works! Thank you.