eelcocramer/node-bluetooth-serial-port #55: Authenticating Pin Code

Globegitter, 2015-01-30:
I just tried it real quick on my Mac and this works beautifully. But it is opening a popup window to enter the pin code. Is there a way to ask directly through the commandline or give it a static pin-code through node?
That would be really useful to have.

eelcocramer, 2015-01-30:
Thanks!

I don't think that is possible on OSX. But you only will be prompted a single time for each device. If possible you can confige your device not to use a code at all.

Globegitter, 2015-01-30:
Aahh thank you for the quick response. I don't suppose you have an idea about windows, do you? The reason I am asking, I am just trying to run this on a device with no screen in the end. So would need something that works full through the terminal.
Just curios if you know anything if that is possible at all. Don't worry if not :)

Thanks for the great library btw!

eelcocramer, 2015-01-30:
I believe on linux it is possible to pair via the commandline but you need to do it before using this module. I don't know about windows, I'm not a user. 

--- sent from the hand

closed 2015-01-30 by eelcocramer

Globegitter, 2015-01-30:
I just got it to work on windows, but it is asking me (like on Mac) for the Pin code. I have some C# code that can bypass the pincode (i.e. entering it programatically). You are definitely positive that there is nothing that can be done?

eelcocramer, 2015-01-31:
Maybe it can be done on windows but on OSX I haven't been able to figure this out. If someone has a pointer on how this can be done I would be happy to add it to the library.