eelcocramer/node-bluetooth-serial-port #56: Sample App

normanb, 2015-02-05:
Hi,

I have tried the sample app with the Android Bluetooth Chat Sample - https://android.googlesource.com/platform/development/+/eclair-passion-release/samples/BluetoothChat/src/com/example/android/BluetoothChat/BluetoothChatService.java

It says it has connected and the data is sent, but it never shows up.

Is there an example of a the library connecting to a server?

Thanks.

eelcocramer, 2015-02-05:
Sorry but I don't really understand your issue.

The node-bluetooth-serial-port library connect to a Bluetooth RFCOMM service on a remote device and enables you to send and receive data from that connection.

Is the Bluetooth Chat Sample you mention a RFCOMM service running on an Android device?

You say that "it says it has connected and the data is sent" but what do you mean by "it"?

There is a sample script in the experiments folder of this project: https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/experiments/scan-test.js

Does this help you?

normanb, 2015-02-05:
'It' is the sample in the README.

The Bluetooth Chat Sample is a RFCOMM service running on Android. 

I have also tried the sample.

How have you tested that this library connects and writes?

eelcocramer, 2015-02-05:
I use to to [connect and control Arduino's](http://servicelab.org/2012/12/12/wirelessly-control-your-arduino-with-nodejs-over-bluetooth/).

So you can see that data was send? Do you see anything in logcat?

Maybe issue #22 can point you in the right direction?

eelcocramer, 2015-05-12:
Closing this for now.

closed 2015-05-12 by eelcocramer