eelcocramer/node-bluetooth-serial-port #57: Leaking Socket / Pipe on Linux?

jeffbk, 2015-02-16:
Hi.  Thank you for providing this really useful library to node js.  I am running on Raspberry PI (Raspbian).  It looks like there may be sockets / pipes leaking.  I am creating a service that periodically attempts to connect to a bluetooth device.  After running for a while, it starts to fail on connect (with an uncaught exception):
```
Error: Cannot create pipe for reading.
    at BluetoothSerialPort.connect (/home/pi/node_modules/bluetooth-serial-port/lib/bluetooth-serial-port.js:81:26)
    at tryConnect (/home/pi/repositories/nodeGateway/test/leak.js:20:14)
    at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)
```
If you run the code below and check the file descriptors (FDs) owned by the process, you will see that each new connection attempt appears to be creating 3 FDs: 1 socket and 2 pipes (using ls -l /proc/<process id>/fd).  Three new FDs are created for each connection attempt and the previous attempt's FDs do not seem to be getting released.  Continue running it until 1024 FDs are consumed to get the error.  I am by no means a linux expert so I might not be interpreting these results correctly.  Perhaps there is something I am doing incorrectly? 
```
var BtSerial = require('bluetooth-serial-port');

var args = process.argv.slice(2);
var bdAddr = args[0];
var channel = args[1];

process.on('uncaughtException', function (error) {
    console.log(error.stack);
    process.exit(1);
});

var btSerial = new BtSerial.BluetoothSerialPort();

function tryConnect() {

    console.log('Connecting...');

    btSerial.connect(bdAddr, channel,
        function () {
            console.log('Connected');
            btSerial.close();
            setTimeout(tryConnect, 1000);
        },
        function () {
            console.log('Connect failed');
            btSerial.close();
            setTimeout(tryConnect, 1000);
        });

}

tryConnect();
```

NOTE: It does not seem to matter if a new BluetoothSerialPort is created for each connection attempt, or if a single BluetoothSerialPort is used for all connection attempts (as shown above).

Thank you again for providing this useful project.

eelcocramer, 2015-02-16:
Thank you for your well documented report. I'll try to look into it somewhere this week.

eelcocramer, 2015-02-17:
In your logging do you see either `Connected` or `Connect failed`? I wonder if `close` gets called correctly.

jeffbk, 2015-02-17:
Thanks for looking into this.  I see the logging output and believe the close is being called in both cases.  I was looking at the linux c code and it appears that two pipes are being open with the socket.  However, it appears that these pipes are not being closed (at least as far as I could tell, not sure if it required either).  If I get a chance I will try to put together a pull request.  It may take me a bit as node modules are new to me and my c coding is a bit rusty...  Thanks.

eelcocramer, 2015-02-18:
Could very well that you pinpointed the issue. In what lines are the pipes created?

jeffbk, 2015-02-18:
It looks like they are created here: https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/linux/BTSerialPortBinding.cc#L269-271
It seems that they may need to be closed according to this example:
http://www.tldp.org/LDP/lpg/node11.html

eelcocramer, 2015-02-18:
I've commited a possible fix. Still need to test this myself but you can give it a try. Checkout branch `issue57`

jeffbk, 2015-02-18:
Unfortunately my RPi is on loan at the moment, but I will test as soon as I get it back (later today).  Sorry for the delay.  Thanks for addressing so quickly.

eelcocramer, 2015-02-18:
No problem. Let me know when you are ready.

eelcocramer, 2015-02-19:
I've just released 1.2.2 with a fix for your issue. Please let me know if you still run into it.

jeffbk, 2015-02-19:
I've done some initial testing and the file descriptors are no longer being leaked (socket and pipes).  Socket and pipes are being cleaned up in both the connect and connect failed cases. :)  
I will do a bit more extensive testing today.  Also, after reviewing the source code a bit more, I am curious as to the necessity of the pipe.  What functionality is it providing?  Thanks.

eelcocramer, 2015-02-23:
Thank you for the feedback.