eelcocramer/node-bluetooth-serial-port #59: could not install on windows box

Mirodil, 2015-02-18:
windows 8.1 x64
```
G:\Projects\NOTE.JS\examples\bluetooth>npm install bluetooth-serial-port
npm WARN package.json socket.io@1.0.0-pre2 No repository field.
npm WARN package.json refresh-token@0.0.2 No repository field.
npm ERR! notsup Unsupported
npm ERR! notsup Not compatible with your operating system or architecture: bluet
ooth-serial-port@1.2.1
npm ERR! notsup Valid OS:    darwin,linux
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   win32
npm ERR! notsup Actual Arch: x64

npm ERR! System Windows_NT 6.2.9200
npm ERR! command "C:\\Program Files\\nodejs\\\\node.exe" "C:\\Program Files\\nod
ejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "bluetooth-serial-port"
npm ERR! cwd G:\Projects\NOTE.JS\examples\bluetooth
npm ERR! node -v v0.10.35
npm ERR! npm -v 1.4.28
npm ERR! code EBADPLATFORM
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     G:\Projects\NOTE.JS\examples\bluetooth\npm-debug.log
npm ERR! not ok code 0

G:\Projects\NOTE.JS\examples\bluetooth>
```

eelcocramer, 2015-02-18:
Currently not supported on the publish version: see #53

You might want to check the current dev version in master branch.

closed 2015-02-18 by eelcocramer

eelcocramer, 2015-02-18:
Will be part of the 1.2.2 release that is scheduled somewhere this week.

eelcocramer, 2015-02-19:
1.2.2 has been released. Please let me know if you still have issues building it.