eelcocramer/node-bluetooth-serial-port #62: Can't connect with an HC-06 Bluetooth module

Eriatolc, 2015-03-19:
Hi everyone,

I'm trying to connect to an HC-06 module connected to my Arduino Yùn using this library. I am writing a program on my computer which will communicate with the Arduino through the Bluetooth HC-06.

The `find()` method of the library does its job very well: it finds the device named "HC-06" with its mac address, and it also finds the channel with the `findSerialPortChannel()` method.

Where things go wrong is when I use the `connect(address, channel, callback)` method, I always have **"can't connect"** as a response.

I don't know where I am doing something wrong...

Here, the code I use to connect:

```javascript
var BTSP = require('bluetooth-serial-port');
var serial = new BTSP.BluetoothSerialPort(); 
 
serial.on('found', function(address, name) {
 	logger.debug('found ', address, ' & name : ', name);
	// you might want to check the found address with the address of your
	// bluetooth enabled Arduino device here.
	if (name === 'HC-06') {
		serial.findSerialPortChannel(address, function(channel) {
			serial.connect(address, channel, function() {
				console.log('connected');
				process.stdin.resume();
				process.stdin.setEncoding('utf8');
				console.log('Press "1" or "0" and "ENTER" to turn on or off the light.')
				 
				process.stdin.on('data', function (data) {
					serial.write(data);
				});
				 
				serial.on('data', function(data) {
					console.log('Received: ' + data);
				});
			}, function () {
					console.log('cannot connect');
			});
		});
	}
});
 
serial.inquire(); 
```

and the output I have:

```bash
$ node bin/arduino.js 
2015-03-18T19:28:17 <debug> arduino.js:21 (BluetoothSerialPort.<anonymous>) found  54:27:1E:AC:0E:A0  & name :  Bluetooth USB Host Controller
2015-03-18T19:28:18 <debug> arduino.js:21 (BluetoothSerialPort.<anonymous>) found  20:13:11:01:26:71  & name :  HC-06
cannot connect

```

eelcocramer, 2015-03-19:
Did you pair the module with the host running the script?

Eriatolc, 2015-03-19:
No, I didn't. Now I did and it works, but I was wondering if there is a way to do the pairing directly using the library?

Eriatolc, 2015-03-19:
More details: now, the connection works, but I can't receive any data from the Serial communication with the Arduino... Any help?

eelcocramer, 2015-03-19:
There is currently no way to pair via this library. Not all platforms support pairing programmatically.

On you second question: are you sure your Arduino is sending data? Can you connect to the device with a serial terminal program (putty on windows or xterm on osx) and receive data?

eelcocramer, 2015-03-24:
@Eriatolc do you still have this issue?

Eriatolc, 2015-03-26:
Hi again.

I have finally resolved my issue. Conclusion: your library is perfect and the Arduino Yùn platform s**ks!

For information, the Serial Communication when you use an HC-06 Bluetooth module on an Arduino Yùn is **not** `Serial` but `Serial1`. Therefore, if you want to write something on the Bluetooth Serial connection, you have to do this:

    Serial1.println("My super test message!");

Concerning the pairing programmatically, I couldn't find any solution yet pairing manually through the GUI of my Ubuntu.

Thanks a lot for your help. I am using your library a lot for my projects.
By the way, last thing, your `connect()` method documentation in the Readme is not corresponding to your examples. In your example, you call this method with two parameters - address and channel - and in your API, it is written that it requires only one parameter (except the callback): the address.

Best regards

closed 2015-03-26 by Eriatolc

mackwic, 2015-03-26:
@Eriatolc If it can reassure you, we also have to teach our customers how to pair with our bluetooth devices in each of their OS because these OSes doesn't  provide any way to do it programatically.

Eriatolc, 2015-03-26:
@mackwic It reassure me a little bit ;) I still find this totally weird that when you are working on embedded systems, especially for a future big production,  you still have to do some things manually... Like it's hard to allow people to send "1234" through the CLI to a Bluetooth device...

eelcocramer, 2015-03-26:
It is possible to pair from the command line [on linux](http://www.heatxsink.com/entry/how-to-pair-a-bluetooth-device-from-command-line-on-linux). Also some OSs can pair automatically but I'm not sure how to set this up. One can argue it is security related to have the user confirm that he wants to connect to the device.

I'll look at the documentation issue you mentioned. Thanks.