eelcocramer/node-bluetooth-serial-port #64: npm install fails on Trisquel/Ubuntu

rodrigopitanga, 2015-03-28:
<code>
rodrigo@tequila4free:~/devel$ npm install bluetooth-serial-port
npm http GET https://registry.npmjs.org/bluetooth-serial-port
npm http 304 https://registry.npmjs.org/bluetooth-serial-port
npm http GET https://registry.npmjs.org/bindings
npm http GET https://registry.npmjs.org/nan
npm http 304 https://registry.npmjs.org/bindings
npm http 304 https://registry.npmjs.org/nan

> bluetooth-serial-port@1.2.2 install /home/rodrigo/devel/node_modules/bluetooth-serial-port
> node-gyp configure build

make: Entrando no diretório `/home/rodrigo/devel/node_modules/bluetooth-serial-port/build'
  CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/BluetoothSerialPort.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/DeviceINQ.o
../src/linux/DeviceINQ.cc:35:37: fatal error: bluetooth/bluetooth.h: Arquivo ou diretório não encontrado
     #include <bluetooth/bluetooth.h>
                                     ^
compilation terminated.
make: ** [Release/obj.target/BluetoothSerialPort/src/linux/DeviceINQ.o] Erro 1
make: Saindo do diretório `/home/rodrigo/devel/node_modules/bluetooth-serial-port/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.EventEmitter.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:797:12)
gyp ERR! System Linux 3.13.0-46-generic
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "configure" "build"
gyp ERR! cwd /home/rodrigo/devel/node_modules/bluetooth-serial-port
gyp ERR! node -v v0.10.25
gyp ERR! node-gyp -v v0.10.10
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! weird error 1
npm ERR! not ok code 0
</code>

Even after installing nodejs-legacy

eelcocramer, 2015-03-28:
Do you have the prerequest installed? Check the readme for instructions. 

It looks like you de not have the bluetooth libs.

rodrigopitanga, 2015-03-29:
that was it! thanks!

closed 2015-03-29 by rodrigopitanga

eelcocramer, 2015-03-30:
Cool. Thanks for updating.