eelcocramer/node-bluetooth-serial-port #65: Unable to install on OSX yosemite

gautaz, 2015-04-07:
Hello,

I'm trying to install bluetooth-serial-port on OSX yosemite, it currently fails.

This seems related to a (not so) recent Apple API modification:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=63651

Here is an excerpt of the npm output (quite long):

10:40 $ LANG=C npm install bluetooth-serial-port
/
> bluetooth-serial-port@1.2.2 install /Users/del/dev/telecom_sante/bed-control-panel/backend/node_modules/bluetooth-serial-port
> node-gyp configure build

  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSObject.h:6:0,
                 from ../src/osx/DeviceINQ.mm:36:
/usr/include/objc/NSObject.h:22:4: error: 'instancetype' does not name a type
 - (instancetype)self;
    ^
/usr/include/objc/NSObject.h:36:4: error: 'instancetype' does not name a type
 - (instancetype)retain OBJC_ARC_UNAVAILABLE;
    ^
/usr/include/objc/NSObject.h:38:4: error: 'instancetype' does not name a type
 - (instancetype)autorelease OBJC_ARC_UNAVAILABLE;
    ^

.... LOTS OF ERRORS AND WARNINGS ....

In file included from /System/Library/Frameworks/CoreServices.framework/Headers/CoreServices.h:55:0,
                 from /System/Library/Frameworks/Foundation.framework/Headers/NSURLError.h:12,
                 from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:80,
                 from /System/Library/Frameworks/IOBluetooth.framework/Headers/IOBluetoothUserLib.h:10,
                 from /System/Library/Frameworks/IOBluetooth.framework/Headers/objc/IOBluetoothDevice.h:9,
                 from ../src/osx/DeviceINQ.mm:37:
/System/Library/Frameworks/CoreServices.framework/Frameworks/FSEvents.framework/Headers/FSEvents.h:414:89: error: expected unqualified-id before '=' token
   kFSEventStreamEventFlagItemCreated __OSX_AVAILABLE_STARTING(__MAC_10_7, __IPHONE_6_0) = 0x00000100,
                                                                                         ^
/System/Library/Frameworks/CoreServices.framework/Frameworks/FSEvents.framework/Headers/FSEvents.h:481:1: error: expected declaration before '}' token
 };
 ^
make: *** [Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/del/.nave/installed/0.10.38/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Darwin 14.1.0
gyp ERR! command "node" "/Users/del/.nave/installed/0.10.38/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "build"
gyp ERR! cwd /Users/del/dev/telecom_sante/bed-control-panel/backend/node_modules/bluetooth-serial-port
gyp ERR! node -v v0.10.38
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok

npm ERR! bluetooth-serial-port@1.2.2 install: `node-gyp configure build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bluetooth-serial-port@1.2.2 install script.
npm ERR! This is most likely a problem with the bluetooth-serial-port package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp configure build
npm ERR! You can get their info via:
npm ERR!     npm owner ls bluetooth-serial-port
npm ERR! There is likely additional logging output above.
npm ERR! System Darwin 14.1.0
npm ERR! command "/Users/del/.nave/installed/ts/bin/node" "/Users/del/.nave/installed/ts/bin/npm" "install" "bluetooth-serial-port"
npm ERR! cwd /Users/del/dev/telecom_sante/bed-control-panel/backend
npm ERR! node -v v0.10.38
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0

eelcocramer, 2015-04-07:
Hi, I've just tried to rebuild the module myself and it build fine.

I'm using an up-to-date OSX Yosemite (10.10.2), XCode (6.2) and nodejs (0.12.2) setup.

gautaz, 2015-04-07:
OK, so the problem is related to my build environment.
I'm kind of reluctant to install the whole Xcode distribution and currently I'm only using the Xcode CLT.
I'm also sometime using gcc 4.9 and I have installed it by using the brew package manager.
I'm also a complete newbie in everything related to node-gyp... (numerous elements are not in my favor :-))
Could you please tell me what is the best way to dig into this and find the root cause of the problem ?

mackwic, 2015-04-07:
Hi @gautaz.
Could you test with `CC=clang npm install bluetooth-serial-port` then with `CC=gcc-4.9 npm install bluetooth-serial-port` ? (Even if CC is historically only for C, I think it will force the compiler also for objective-C, tell me if it doesn't. Maybe `CC='gcc-4.9 -x objective-c'` ?)

I suspect that
1) `LANG=C` break the compiler suite. Test without this variable
2) it will compile better with `CC=clang` than `CC=gcc-4.9`, but it's only a suspicion.

Tell us the results, please.

gautaz, 2015-04-07:
Hi @mackwic,

I have tried both commands with the same results.
It seems that the npm/node-gyp process do not take into account the environment, I have tried this:
`CC=blub npm install bluetooth-serial-port`
and the result was still the same (same errors and obviously the `blub` command does not exist).

Concerning `LANG=C`, this has just some effect on the current locale used by the system (I wanted the error messages to be displayed in english), removing it still produces the same errors (which are then just localized to my native language).

eelcocramer, 2015-04-07:
The easy check might be to temporarily remove gcc (with brew unlink or something). Or remove `/usr/local/bin` from your PATH when you execute `npm install` to see if gcc (or something that was installed by homebrew) is causing the issue.

gautaz, 2015-04-07:
Thanks @eelcocramer,

This indeed does it as the upstream gcc does not currently support the Apple API change as stated in https://gcc.gnu.org/bugzilla/show_bug.cgi?id=63651.

For now, I can use a modified PATH when calling npm:
`PATH="yadayada" npm install bluetooth-serial-port`

It would be even better if I could directly tell node-gyp which compiler chain it has to use...
Anyway, I have a workaround for now.

closed 2015-04-07 by gautaz

eelcocramer, 2015-04-07:
Cool you can work around the issue for now. Maybe you can check with the node-gyp guys if there is a more permanent solution for your case.

Thanks for getting back!