eelcocramer/node-bluetooth-serial-port #66: What is the behavior on power loss ?

gautaz, 2015-04-08:
Hello,

I am wondering what should be the behavior of the bluetooth-serial-port library when a device it is already paired with suffers a power loss.
From my experiments, the library returns an error on writing ("Write was unsuccessful") but is there a way to be more proactive, i.e. listening on a particular event emitted by the library ?
It seems that neither "closed" nor "failure" are emitted in this particular case...

eelcocramer, 2015-04-08:
Unfortunately, as far as I know, there is no way to do that. What the library does is sending the event as soon as the OS signals the library. If there is no event from the OS there is nothing to send upwards.

You could implement your own `keep-alive` messages if you want...

gautaz, 2015-04-08:
Thanks @eelcocramer for the `keep-alive` idea, I'll try this for now.

eelcocramer, 2015-05-12:
Adding this to the FAQ for future reference.

closed 2015-05-12 by eelcocramer