eelcocramer/node-bluetooth-serial-port #69: How to make the bluetooth service visible

alanbly, 2015-05-16:
I'm trying to build a bluetooth serial bridge between two Macs using this library, but I can't seem to figure out how to make them talk to each other. I've tried just manually picking a channel (which I presume the OS is stopping) and trying to have each one scan the other for serial ports which lists the same services whether the other side is running or not. I'm wondering if that's just not built into this library right now or if I'v got something wrong in my setup. 

My code for each side is [here](https://github.com/alanbly/bluetooth-bit-stream) if you're interested

eelcocramer, 2015-05-16:
Sorry but this library is client side only. If you have BT serial port service running somewhere you can use this library to connect to it. I'm unaware of any node modules that implement the server side.

alanbly, 2015-05-16:
Thanks for the quick response. I'll just have to build it.

eelcocramer, 2015-05-18:
Cool. Consider open sourcing it :+1:

I'll close the issue for now.

closed 2015-05-18 by eelcocramer

eelcocramer, 2016-09-13:
As from version 1.3.1 this library has server support on Linux, see also #103