eelcocramer/node-bluetooth-serial-port #70: Unable to open a socket between Android and osx 10.9

mackwic, 2015-05-28:
Hi @eelcocramer !

So, we had to transfer data between our cross-platform desktop application and our Android application (to be precise: records from a wearable product).

Here is the [node code](https://gist.github.com/mackwic/c46a812cfcff30400b0f) and here is the [android code](https://gist.github.com/Footjy/9bc443eccb830ff9be25) (both stripped, of course).
For reference, it's a variation on the standard Android [Bluetooth Chat Sample](https://android.googlesource.com/platform/development/+/eclair-passion-release/samples/BluetoothChat/src/com/example/android/BluetoothChat/BluetoothChatService.java).

On linux, the connection opens and we can transfer all the data we want repeatably.
On osx 10.9, the connection opens on the osx side but not on the android app side. We can use `btSerial.write()` but as the socket is not opened on both sides, data is never received.
We tested with 3 different Android devices on 4.4 and 5.0. No connection has been opened.

That said, we use bluetooth-serial-port to connect with the wearable device and stream data at high rate with no issues, on all platforms, so it's definitely a tricky bug involving implementation details between osx and android bluetooth stack.

I don't know what information to add. As I said, I tried to recompile with different obvious tweaks (like using the SPP uuid rather than RFCOMM) with no success. Any idea ?

eelcocramer, 2015-05-28:
Thanks. I do not have an Android device available for the coming few weeks to test this out myself.

For sanity, can you try connecting a terminal program to connect to your Android service from a Mac to check if that opens up the connection ok? I've used [zterm](http://www.dalverson.com/zterm/) to do this.

Just for reference: do you use OSX Yosemite?

mackwic, 2015-05-29:
We tested as you recommended, and the result are troublesome:
1. ZTerm need the tty to be already open, so we used bluetooth-serial-port to open this tty (knowing that it will fail)
2. Surprise ! It happened to work. Well, once or two. After de-pairing both sides, disabling/enabling bluetooth both sides, and 2 use of bluetooth-serial-port (the first fail _as expected_), the second try opened correctly the socket. We succeed with this protocol two times.
3. When the tty exists, ZTerm _always_ open correctly the socket on both sides. Which reinforce the idea of a miss-configuration of the socket on bluetooth-serial-port side in a special edge case (but doesn't explain how it could have worked)

About Yosemite: _I_ don't use it personally, but my clients does so obviously I'll need to test carefully with it. Didn't have time but I will soon

eelcocramer, 2015-05-29:
Hmmm. I hope you can nail this to a piece of code.

I could always open the port with ZTerm as long as my mac was paired with the device and the services where discovered ok using the Bluetooth Setup Assistant.

mackwic, 2015-05-29:
Code is unchanged, same as first comment. What can I add ?

eelcocramer, 2015-05-29:
Looking at the java code I see that the bluetooth chat example has all the outer class methods `synchronized` where you did not do that. Not sure at all if this might be a problem, it was just something I noticed.

Where in the Android code is the code blocking when the connection is not opened? On the `mmServerSocket.accept();`?

Footjy, 2015-05-29:
Hi, please let me answer as I wrote the application.

The synchronized key word is about threading, it can't be related to the bug, but thanks for notifying it as it is still good practice to use it in that case (it makes all the lines of the synchronized methods executed together, preventing the STATE variable from being changed by another thread in the middle of the execution of the method).

It is indeed on the    mmServerSocket.accept();    that the program blocks.

eelcocramer, 2015-05-29:
I know what the `synchronized` keyword does. I mentioned it since you are getting inconsistent results (it sometimes does work right?). Although it seems that the problem most likely lies within OSX code of this module I do not want to rule out the Android side yet ;-)

Footjy, 2015-05-29:
Better be too specific than not enough ^^

I have been working a lot with bluetooth on Android. I would not be sure that the Android side does not have any mistake, but for sure synchronized is not the point :)

Thanks for pointing this out.

eelcocramer, 2015-07-06:
@mackwic did your previous PRs fix this issue as well? Or it this still relevant?

mackwic, 2015-07-06:
I don't think so. We didn't had time to test it further, though, as we moved to another solution WiFi based.

I would be glad to make some tests if you need to.

closed 2016-04-28 by eelcocramer