eelcocramer/node-bluetooth-serial-port #71: Large writes fail

djMax, 2015-06-02:
It seems that large writes fail. First, the length looks like it's a UInt16. But also the send callback gets called many times - often with -1 for the second arg. What's happening here? What does -1 mean and why does it recover "gracefully"?

eelcocramer, 2015-06-02:
Not an issue or did you resolve it?

djMax, 2015-06-02:
Not an issue - was in the USB version of the code, so it's a bug with node-serialport not bt.

djMax, 2015-06-03:
Well, it turns out this might be a problem in nbsp too... If I write small packets, say 8192 bytes and then wait for a response, I get one. If I write large amounts of data, even writing say 8k at a time and waiting for the write callback to be fired, then the device never receives the data. It's as if the buffering is deadlocking or something. (Mac OS X)

djMax, 2015-06-03:
It looks like it's getting kIOReturnNoSpace back from BT.

djMax, 2015-06-03:
Also, delaying 100msec between writes of 8k does work.

eelcocramer, 2015-06-03:
So if you write 8k and then delay you will receive a callback? Sounds like you overflowing the buffer?

In the API docs of `IOBluetoothRFCOMMChannel` is only described what the success case return code of `writeAsync` is. The `kIOReturnNoSpace` isn't described there. Sounds like it is a return code for when the buffers or remote device is not accepting new data at that specific time.

What would you expect the module to do when this happens?

djMax, 2015-06-03:
It's very odd... I put in retry logic that says "if you get an error on write, wait 100 msec and try again." But this is wildly slower than always waiting 100msec all the time. Anyhow, I think the concrete change is that right now the node layer doesn't get an error code back on a write failure, and it probably should. "Out of space" seems platform agnostic enough that it could be surfaced as that.

eelcocramer, 2015-06-04:
I gave this some thought and I believe that just kicking the failure upstream is not good enough. Writing to the BT device happens in a loop and when the buffer that is written is larger than the MTU there is no way to know upstream what data is and isn't written correctly. I think it is better so solve this by either:

1. Try the write again when it returns a `kIOReturnNoSpace`.
2. Do not allow the buffer to exceed the MTU size.

The problem that I have with option 1 is that the return codes are not really documented. I do not like to assume that when this happens the module can recover from the situation by itself.

I would prefer option 2 but it would require an API change on all platforms to expose the MTU to the outside of the module.

Any thoughts on this?

eelcocramer, 2015-06-04:
Actually, look at the code I would have expected that in the case the `kIOReturnNoSpace` occurs the callback function gets called with an error as the first argument. The error should have a `Write was unsuccessful` message in it.

djMax, 2015-06-04:
It does, but the higher level has no idea why, so it's not generally safe to assume a delayed retry will work, I think.

djMax, 2015-06-04:
I think making the higher level manage the MTU is unpleasant. The retry option seems fine and easy in JS, but the upper layer just needs more information to know that's what it should do.

eelcocramer, 2015-06-04:
So I made a small change to the code which will hopefully fix your issue. But as a currently away from my BT test devices I cannot test it myself so I currently have no clue if this will work at all.

I changed the callback to the write so that when the first `Error` argument is defined (in case of an error) the 2nd argument is also passed. This arguments holds the number of bytes that were written successfully.

So when you receive the callback and the error is set you can check if any bytes of your buffer where written at all. In the next attempt you should skip the first X bytes of your buffer as they are already written.

I've only changed this in the OSX code for now.

Are you willing to check if this works for you? You can find the changes in the `issue71` branch.

eelcocramer, 2015-06-14:
Did you have a change to test this branch?

closed 2016-04-28 by eelcocramer