eelcocramer/node-bluetooth-serial-port #72: how to install this module in mac

trvajjala, 2015-06-15:
im trying to install  using following command in mac

$/> sudo npm install -g "node-bluetooth-serial-port"

'node-bluetooth-serial-port' is not in the npm registry.

where im doing wrong

eelcocramer, 2015-06-15:
npm install bluetooth-serial-port

trvajjala, 2015-06-15:
here is the  error console

sudo npm -g install bluetooth-serial-port
Password:
-
> bluetooth-serial-port@1.2.3 install /usr/local/lib/node_modules/bluetooth-serial-port
> node-gyp configure build

module.js:338
    throw err;
          ^
Error: Cannot find module '/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Function.Module.runMain (module.js:501:10)
    at startup (node.js:129:16)
    at node.js:814:3
npm ERR! Darwin 14.1.0
npm ERR! argv "node" "/usr/local/bin/npm" "-g" "install" "bluetooth-serial-port"
npm ERR! node v0.12.3
npm ERR! npm  v2.11.2
npm ERR! code ELIFECYCLE

npm ERR! bluetooth-serial-port@1.2.3 install: `node-gyp configure build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bluetooth-serial-port@1.2.3 install script 'node-gyp configure build'.
npm ERR! This is most likely a problem with the bluetooth-serial-port package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp configure build
npm ERR! You can get their info via:
npm ERR!     npm owner ls bluetooth-serial-port
npm ERR! There is likely additional logging output above.

eelcocramer, 2015-06-15:
Check your local setup. It looks like you node and npm versions do not match. Node v0.12.3 should have installed npm v2.9.1.

trvajjala, 2015-06-15:
ras:~ tvajjala$ npm -v
2.11.2
ras:~ tvajjala$ node -v
v0.12.3

eelcocramer, 2015-06-15:
Could very well be. If I google on your error (to do with nod-gyp) I find problems that are caused with incompatible versions.

mackwic, 2015-06-17:
Seeing the issue, are you sure you have installed `node-gyp`, which is required to compile native modules (like bluetooth-serial-port) ?

closed 2016-04-28 by eelcocramer