eelcocramer/node-bluetooth-serial-port #73: Module version mismatch

gtzogana, 2015-06-15:
I get this error when I try to run any code requiring `bluetooth-serial-port` in Github's Electron environment (Chromium):

A JavaScript error occured in the browser process
Uncaught Exception:
Error: Module version mismatch. Expected 43, got 14.
    at Error (native)
    at Object.module.(anonymous function) (ATOM_SHELL_ASAR.js:134:20)
    at Object.module.(anonymous function) [as .node] (ATOM_SHELL_ASAR.js:134:20)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:298:12)
    at Module.require (module.js:353:17)
    at require (module.js:372:17)
    at bindings (/home/george/Desktop/turntable/phalanx/node_modules/bluetooth-serial-port/node_modules/bindings/bindings.js:76:44)
    at /home/george/Desktop/turntable/phalanx/node_modules/bluetooth-serial-port/lib/bluetooth-serial-port.js:20:39
    at Object.<anonymous> (/home/george/Desktop/turntable/phalanx/node_modules/bluetooth-serial-port/lib/bluetooth-serial-port.js:131:2)

Is this module unsupported in Electron?

eelcocramer, 2015-06-15:
I've never tried or heard of Electron. But you might find this post on stackoverflow useful: http://stackoverflow.com/questions/28771245/module-version-mismatch-with-headless-browser-and-desktop-node

Can you let me kniw if this helps?

gtzogana, 2015-06-15:
That did it. Thanks!

eelcocramer, 2015-06-15:
Thank you for the feedback.

closed 2015-06-15 by eelcocramer

amrish7, 2016-09-28:
@gtzogana Wondering if you were able to get `bluetooth-serial-port` work in Electron? Please take a look at https://github.com/eelcocramer/node-bluetooth-serial-port/issues/116 and give your thought? I am having trouble connecting to a paired device