eelcocramer/node-bluetooth-serial-port #79: finds but isn't connected

versoul, 2015-08-07:
Hello I have such problem, at start of an example it is possible to find pair of devices, but connection to them doesn't even begin = (

eelcocramer, 2015-08-07:
Can you provide more details? For example your code and logging output? 

Thanks.

versoul, 2015-08-07:
code
```
var BluetoothSerialPort = require("../lib/bluetooth-serial-port.js").BluetoothSerialPort;
var myBtPort = new BluetoothSerialPort();
var dataBuffer = "";

myBtPort.on('found', function (address, name) {
    console.log('Found: ' + address + ' with name ' + name);

    myBtPort.findSerialPortChannel(address, function(channel) {
        console.log('Found RFCOMM channel for serial port on ' + name + ': ' + channel);

        console.log('Attempting to connect...');

        myBtPort.connect(address, channel, function() {
            console.log('Connected. Receiving data...');

            myBtPort.on('data', function(buffer) {

                dataBuffer = dataBuffer + buffer.toString('utf-8');

                if(dataBuffer.indexOf("\n") != -1){
                     console.log("New incoming string");
                    console.log("String = " + dataBuffer);
                    dataBuffer = "";
                }

            });
        });
    });
});

myBtPort.on('finished', function() {
    console.log('scan did finish');
});

myBtPort.inquire();
```

the answer 2 devices are found, and further to them doesn't try to connect though bluetooth works regularly

```
versoul@versoul-laptop:~/workspace/racestory/tst$ hcitool scan
Scanning ...
    44:74:6C:8D:DC:24   Xperia Z3
versoul@versoul-laptop:~/workspace/racestory/tst$ sudo l2ping -c 1 44:74:6c:8d:dc:24
Ping: 44:74:6c:8d:dc:24 from 00:11:67:BA:03:2D (data size 44) ...
0 bytes from 44:74:6c:8d:dc:24 id 0 time 12.75ms
1 sent, 1 received, 0% loss
```

eelcocramer, 2015-08-07:
Do you get output like:

```
Found RFCOMM channel for serial port on <your device>: <your channel>
Attempting to connect...
```

What OS are you using?
Is you device already paired with your laptop?

versoul, 2015-08-07:
i use ubuntu and raspbian, no dont paired
how i can do this from terminal?

versoul, 2015-08-07:
i have only yhis log
console.log('Found: ' + address + ' with name ' + name);
next function dont call

eelcocramer, 2015-08-07:
You should pair the device before connecting to it with this module. I've done this using [bluez on Ubuntu](http://askubuntu.com/questions/187092/bluetooth-from-the-command-line-in-12-04). I hope this helps.

versoul, 2015-08-07:
its work, thanks
but on raspbian install with errors, but works...
```
pi@raspberrypi ~/racestory/tst $ sudo npm install bluetooth-serial-port
\
> bluetooth-serial-port@1.2.4 install /home/pi/racestory/node_modules/bluetooth-serial-port
> node-gyp configure build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.10.38"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/racestory/node_modules/bluetooth-serial-port/.node-gyp"
make: вход в каталог «/home/pi/racestory/node_modules/bluetooth-serial-port/build»
  CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/BluetoothSerialPort.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/DeviceINQ.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/linux/BTSerialPortBinding.o
../src/linux/BTSerialPortBinding.cc: In static member function ‘static void BTSerialPortBinding::EIO_Write(uv_work_t*)’:
../src/linux/BTSerialPortBinding.cc:113:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (data->result != data->bufferLength) {
                      ^
../src/linux/BTSerialPortBinding.cc: In static member function ‘static v8::Handle<v8::Value> BTSerialPortBinding::New(const v8::Arguments&)’:
../src/linux/BTSerialPortBinding.cc:262:21: warning: variable ‘cb’ set but not used [-Wunused-but-set-variable]
     Local<Function> cb = Local<Function>::Cast(args[2]);
                     ^
../src/linux/BTSerialPortBinding.cc:263:21: warning: variable ‘ecb’ set but not used [-Wunused-but-set-variable]
     Local<Function> ecb = Local<Function>::Cast(args[3]);
                     ^
  SOLINK_MODULE(target) Release/obj.target/BluetoothSerialPort.node
  SOLINK_MODULE(target) Release/obj.target/BluetoothSerialPort.node: Finished
  COPY Release/BluetoothSerialPort.node
make: выход из каталога «/home/pi/racestory/node_modules/bluetooth-serial-port/build»
bluetooth-serial-port@1.2.4 ../node_modules/bluetooth-serial-port
├── bindings@1.1.1
└── nan@1.9.0
pi@raspberrypi ~/racestory/tst $ sudo node btserial.js 
Found: 00:12:05:16:99:52 with name linvor
scan did finish
Found RFCOMM channel for serial port on linvor: 1
Attempting to connect...
Connected. Receiving data...

```

eelcocramer, 2015-08-07:
Cool. Thanks for the feedback.

You should be able to build, run and connect to remote devices as an ordinary user. Using `root` should not be needed (for this module anyway).

Happy building!

closed 2015-08-07 by eelcocramer