eelcocramer/node-bluetooth-serial-port #83: Runtime error with node 4

jeroenhouben, 2015-10-22:
Hi,

This package used to compile & run fine but since node 4 its throwing this error

```
FATAL ERROR: v8::HandleScope::CreateHandle() Cannot create a handle without a HandleScope
Abort trap: 6
```

more info: https://github.com/nodejs/nan/issues/440

eelcocramer, 2015-10-22:
Thanks for creating the issue. Is it a runtime or compile time issue?

Can you give some info about your setup (os, arch, etc) and about when the error occurs?

jeroenhouben, 2015-10-22:
runtime.

both osx 10.11 and arm7

eelcocramer, 2015-10-22:
Thanks. Does it happen on a specific api call? Initial quick check of the osx code looks fine with regard to the issue you are refering. 

Will look into it further over the weekend.

jeroenhouben, 2015-10-23:
My program paires with the remote slave, but then this function seems to blow up, the callback is never fired.

```
btSerial.findSerialPortChannel()
```

WebWalks, 2015-10-23:
Hi, i'm experiencing the same problem - downloaded the packages from npm this afternoon and hit the same issue.
I initially tried to connect to a bluetooth device from OSX 10.11 (El Capitan), then from a raspberry pi, and both got the same error.
I'm running Node v4.1.2, and (using your example code) it finds the address and name ok (on found), but fails while finding the serial port channel
```
btSerial.on('found', function(address, name) {
    btSerial.findSerialPortChannel(address, function(channel) { //FAILS ON THIS CALL
```

During the build (on OSX) i get 2 warnings:

```
node-gyp configure build

  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/BluetoothWorker.o
../src/osx/BluetoothWorker.mm:117:39: warning: null passed to a callee that requires a non-null argument [-Wnonnull]
    keepAliveTimer = [[NSTimer alloc] initWithFireDate:[NSDate distantFuture]
                                      ^
../src/osx/BluetoothWorker.mm:117:39: warning: null passed to a callee that requires a non-null argument [-Wnonnull]
    keepAliveTimer = [[NSTimer alloc] initWithFireDate:[NSDate distantFuture]
                                      ^
2 warnings generated.
```

Not sure if these have any bearing on the problem, but thought i would include them.

k.

eelcocramer, 2015-10-23:
Thanks for the report.

@jeroenhouben are you on node v4.1.2 as well?

To be able to resolve this quickly I would really appreciate a stacktrace to see in exactly what line of the native code the error occurs.

I'll test this myself after the weekend as I've no device to test with atm.

jeroenhouben, 2015-10-24:
yes. 4.1.2

I don't know how to get a stacktrace. This is all I see:

```
found HC-05 on 30-14-10-09-12-12 // my own console.log()
FATAL ERROR: v8::HandleScope::CreateHandle() Cannot create a handle without a HandleScope
Abort trap: 6j
```

eelcocramer, 2015-10-24:
Ok thanks, I will try after the weekend...

Can one of you guys try to update to the latest version of node (4.2.1) to see if this issue also occurs on that version?

WebWalks, 2015-10-25:
Sorry, was offline yesterday..

I just popped into the office to try v4.2.1 but i get exactly the same issue (again, no stack trace is shown - is there something i can do to get the stack trace??)

I also tried v4.0.0 but same error!

Trying v0.10.40, v0.11.16, and v0.12.7 I DONT get the error, but it never seems to find a channel (perhaps thats a different issue)

I can use v0.12 for now to try and diagnose why i'm not getting a channel, but getting it on latest node version would be great.

Thanks for your help on this @eelcocramer !

eelcocramer, 2015-10-25:
Thank you for the feedback. I will look into this on Monday.

eelcocramer, 2015-10-26:
I can reproduce this. I will look into it further and keep you updated.

eelcocramer, 2015-10-26:
I've pinned this issue down and also have a possible fix for it. But the same error occurs on other places as well but the fix is not applicable on all places so I really need to dive into these handle scopes again.

The call stack for the error is below:

```
* thread #1: tid = 0x5c1c6, 0x00007fff943f00ae libsystem_kernel.dylib`__pthread_kill + 10, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff943f00ae libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff82281500 libsystem_pthread.dylib`pthread_kill + 90
    frame #2: 0x00007fff9510c37b libsystem_c.dylib`abort + 129
    frame #3: 0x00000001004eb107 node`node::OnFatalError(char const*, char const*) + 73
    frame #4: 0x00000001001301d9 node`v8::Utils::ReportApiFailure(char const*, char const*) + 57
    frame #5: 0x000000010026e0d1 node`v8::internal::HandleScope::Extend(v8::internal::Isolate*) + 113
    frame #6: 0x0000000100148907 node`v8::Integer::New(v8::Isolate*, int) + 41
    frame #7: 0x00000001010cdcdb BluetoothSerialPort.node`Nan::imp::IntegerFactory<v8::Int32>::New(value=1) + 27 at nan_implementation_12_inl.h:158
    frame #8: 0x00000001010cdca3 BluetoothSerialPort.node`Nan::imp::Factory<v8::Int32>::return_t Nan::New<v8::Int32, int>(arg0=1) + 19 at nan_new.h:208
    frame #9: 0x00000001010ccca3 BluetoothSerialPort.node`Nan::New(value=1) + 19 at nan_new.h:275
    frame #10: 0x00000001010caec6 BluetoothSerialPort.node`DeviceINQ::EIO_AfterSdpSearch(req=0x0000000101405c48) + 70 at DeviceINQ.mm:64
    frame #11: 0x00000001006103b2 node`uv__work_done + 175
    frame #12: 0x0000000100611e96 node`uv__async_event + 65
    frame #13: 0x0000000100612019 node`uv__async_io + 136
    frame #14: 0x000000010062029d node`uv__io_poll + 1520
    frame #15: 0x00000001006124cb node`uv_run + 321
    frame #16: 0x00000001004f0b2f node`node::Start(int, char**) + 557
    frame #17: 0x0000000100002834 node`start + 52
(lldb) q
```

jeroenhouben, 2015-10-26:
I cannot help you there, I know abs nothing about C extensions.

Good luck tho! :)

eelcocramer, 2015-10-27:
If you're up to it you can checkout the 'issue83' branch and build and test that. My simple test works fine. I'm not ready to publish it as I don't feel completely comfortable yet with the use of scopes (especially in the read function). I still need to dive into the details and understand how this is suppose to work and why it is not working in the master branch.

jeroenhouben, 2015-10-28:
:+1: 

This fixes things for me! Thanks!!

eelcocramer, 2015-10-29:
Thanks! If everything goes well U will push a new version to npm after the weekend.

jeroenhouben, 2015-10-29:
OK I seem to have some different problems under ARM7 still, but that could
well be in my own code, so I won't report that just yet :)

eelcocramer, 2015-11-02:
Published version 1.2.6 to npm.

commit 9116ea5e (2015-10-29): Fixes #83.

Have to make sure that this scoping stuff is correct though...
closed 2015-11-02 by eelcocramer