eelcocramer/node-bluetooth-serial-port #85: Can't connect on OSX

Niksko, 2015-11-03:
Hi

I'm running into an issue with connection to devices on OSX. This is both on a roughly 3 year old Macbook Air running OSX Yosemite, and on a one year old Macbook pro running OSX El Capitan.

The issue is that in order to successfully connect to a device via serial, I have to perform the following steps:

* Open Bluetooth preferences
* Right click on the device and click 'remove'
* Wait for the Mac to rediscover the device
* Pair with the device

This sequence needs to be repeated after every successful connection to the device. Not removing and re-pairing with the device gives a:
```
[Error: Cannot connect]

Error: Cannot connect
    at Error: native
```

This also appears to be related to scanning for serial devices using `inquire()` as opposed to just accessing the devices via the list of paired devices. I wasn't able to connect to the device when I was discovering devices using inquiry, but once I stopped performing the inquiry and started using the paired list it work.

Additionally, inquiry and connection works flawlessly on my Linux machine running Arch Linux on kernel 4.0.7.2 (not the latest kernels, because there appears to be a Bluetooth bug in versions 4.1 and 4.2).

Please note that this is using Node.js version 0.12.7, because one of the libraries I use doesn't support version 4 or 5.

eelcocramer, 2015-11-03:
What version of the module are you using?

Have you always had this problem on OSX or is this a regression?

Niksko, 2015-11-05:
I've never used this module prior to this, so I can't say if it's a regression or not.

I *think* this on 1.2.6, but I can't be sure because it was on a colleague's machine. It says you published the update 3 days ago, and I'm fairly sure I updated my dependencies earlier than that.

eelcocramer, 2015-11-05:
Ok thanks. I will see if I can reproduce this with node 0.12.7

eelcocramer, 2015-11-10:
I've tested with node 0.12.7 but cannot reproduce this. You issue sounds familiar though as I remember that I've been in a similar situation some time ago. However I do not remember how I resolved it eventually all tough I've been having BT issues with my MacBook and did all kinds of stuff to get them solved.

Niksko, 2015-11-11:
That's alright. It seems like we've come up with a standard procedure for getting the connection to work, as I said above.

Not sure if this helps, but I've been noticing in the last few days that if I do what I've outlined above (remove the device from the paired list, wait for rediscovery, pair), it sometimes doesn't work. I'm reasonably sure this is time sensitive, in that if I do re-pairing procedure and then connect immediately, it will connect. Where as if I wait maybe 2 or 3 minutes, it wont.

Could it be related to the fact that OSX seems to like to connect as soon as I've paired? I *think* I've tried immediately disconnecting after the pair, but that seems to give the error. I wish the error were more descriptive. Would some sort of log help? I know on Linux you can do a `hcidump` to view the raw bluetooth packet data. Is there something similar on OSX?

eelcocramer, 2015-11-11:
You might want to checkout the [Hardware IO Tools](http://stackoverflow.com/questions/5863088/bluetooth-sniffer-preferably-mac-osx) for XCode.

vincentsong, 2016-01-28:
I have one pretty same issue. But the problem is that I have a device which support normal RFComm Serial port and iAP protocol which is based on RFComm too. e.g. 

   5 RFComm Channel
   6 iAP Channel

When I call findSerialPortChannel it always return 6 instead of 5, and it will always failed to connect channel 6, but it will be OK if I connect to channel 5.

do you know how can I get channel 5 instead of 6?

vincentsong, 2016-01-29:
Hi @eelcocramer , I have another question, what IDE do you guys use to develop Nodejs Addons like this project? Or you guys just use gyp command line? I am on Mac OSX. Really want to contribute something but don't know where to start.  :D

eelcocramer, 2016-01-29:
Hi Vincent,

Thanks for responding here. You issue does not sound familiair to me. Does the the module return both channels or only the iAP channel?

As for developing native modules: I'm not aware of an IDE for native node modules. I'm using VIM with some nice addon modules for all my development work.

Good luck!

closed 2016-04-28 by eelcocramer