eelcocramer/node-bluetooth-serial-port #86: bluetooth with multiple dongles

momohuri, 2015-11-04:
Hi,
Quick question.
I am planning to buy a multiple bluetooth dongles. And I want to connect each of them to a different device and stream different data to each of them.
Will it be possible using this module ? Or is there a problem that I didn't anticipated?

thanks

eelcocramer, 2015-11-05:
Hi,

You can use this module to discover and connect to other bluetooth devices that support the bluetooth serial port profile. On you devices you will need software that listens to incoming connections and handle those connection. You cannot use this software for that task.

* This module is client side only.
* This module only supports the Bluetooth serial port profile.

I hope this helps.

Niksko, 2015-11-05:
I don't think you've really answered his question. I'm actually interested in this as well, though my use case is slightly different and I think I have a workaround regardless of your answer.

My reading of what @momohuri is asking is this: If I have multiple bluetooth controllers on one host device, can I use this library to connect to multiple client devices over serial? There doesn't seem to be any capability in this library to select which bluetooth controller to use, which makes me wonder if perhaps it isn't possible.

That being said, I just assumed that a bluetooth controller could only connect via serial to one client device at a time. Is this correct?

eelcocramer, 2015-11-05:
It should be possible to use this module to setup connections to multiple devices at the same time (up to 8?). That said, I've never tried this myself.

eelcocramer, 2015-11-05:
As for your other question: I don't think the bluetooth stack for OSX and Linux support multiple bluetooth controllers in the first place? Don't know about Windows.

momohuri, 2015-11-05:
Hi,
@Niksko you got my question correctly. Thanks for explaining it. I am not English native speaker...
@eelcocramer I plan to use linux, I was also wondering about the native support. 
Thanks for you answer. It might try to find an other solution.

Niksko, 2015-11-06:
@eelcocramer Honestly, I haven't tried yet, though I will be soon. However on Linux, lots of bluetooth diagnostic commands (`hictool`, `hciconfig`, `hcidump`) all have an option for specifying which controller you want to use, so that makes me think it's supported.

closed 2016-04-28 by eelcocramer