eelcocramer/node-bluetooth-serial-port #87: Segmentation fault: 11

StritziK, 2015-11-09:
First, excellent work!
I'm using a cronJob to connect to my bluetooth device every minute. Therefore I inquire a connection, connect and close it 5 sec after writing. Also connecting and sending for a second time works fine but when I try to close the connection my node server crashes and it only says 
```console
Segmentation fault: 11
```
Here is the snippet am I doing something wrong?:
```javascript

var serial = new BluetoothSerialPort();

function BtConnector() {}

BtConnector.prototype.exec = function(tCommand) {
        command = tCommand;
        serial.inquire();
    };

    serial.on('found', function(address, name) {
        serial.findSerialPortChannel(address, function(channel) {
            console.log('Found RFCOMM channel for serial port on ' + name + ': ' + channel + ': ' + address);
            if (address !== sysAddress) {
                return;
            }
            console.log('Attempting to connect...');
            serial.connect(address, channel, function() {
                var buf = new Buffer(command);
                serial.write(buf, function(err, count) {
                    if (err) {
                        console.log('Error received: ' + err);
                    } else {
                        console.log('Bytes written is: ' + count);
                    }
                    setTimeout(function() {
                        serial.close();
                        console.log('Closed and ready');
                    }, 5000);
                });

            }, function() {
                console.log('cannot connect');
            });
        }, function() {
            console.log('found nothing');
        });
    });

    serial.on('data', function(buffer) {
        console.log(buffer.toString('utf-8'));
    });

    serial.on('finished', function() {
        console.log('scan did finish');
    });
```

eelcocramer, 2015-11-10:
Thanks for your report. I will try to reproduce it.

eelcocramer, 2015-11-10:
I cannot reproduce this with the above code but segfaults are nasty.

What you could do is try to debug this yourself. To do so you need to recompile this module with debug flags by going to `cd node_modules/node-bluetooth-serial-port` and then `node-gyp clean configure build --debug`.

After that you can start your script in the debugger: `lldb -- node <your script>`. After this command you see the prompt of the debugger. You can start your script with the command 'r' and when the segfault happens you can print a stack trace using the command 'bt'. If you can post the output of `bt` here I will take a look at it.

eelcocramer, 2015-11-10:
Also I noticed that you do not define a var `command` in your script but use the value from the scope of `exec`. Javascript is pretty forgiving but it would be better to put all the `serial.on` stuff within the `exec` function so it is in the same scope as the `command` var. It is also good practice to [explicitly define your variables](https://jslinterrors.com/a-was-used-before-it-was-defined).

StritziK, 2015-11-10:
Thanks for your response, my fault - i posted only a excerpt of my code. Now I updatet it. Here's the text it printed with command bt.
```console
(lldb) bt
* thread #8: tid = 0x188851, 0x00007fff8820448d libsystem_pthread.dylib`pthread_mutex_lock, stop reason = EXC_BAD_ACCESS (code=1, address=0x88)
  * frame #0: 0x00007fff8820448d libsystem_pthread.dylib`pthread_mutex_lock
    frame #1: 0x00000001023cd824 BluetoothSerialPort.node`pipe_consumer_free(handle=0x0000000000000000) + 36 at pipe.c:671
    frame #2: 0x00000001023cff2a BluetoothSerialPort.node`BTSerialPortBinding::EIO_Read(req=0x00000001010af008) + 234 at BTSerialPortBinding.mm:165
    frame #3: 0x000000010048c666 node`worker + 90
    frame #4: 0x000000010049756e node`uv__thread_start + 25
    frame #5: 0x00007fff882069b1 libsystem_pthread.dylib`_pthread_body + 131
    frame #6: 0x00007fff8820692e libsystem_pthread.dylib`_pthread_start + 168
    frame #7: 0x00007fff88204385 libsystem_pthread.dylib`thread_start + 13
```

eelcocramer, 2015-11-10:
Thanks for the backtrace and setting the javascript right 😉

I've pushed a new branch called `issue87` which might fix your issue. After you called `close`, the `read` function returns and there is some code to free memory allocated to buffers but apparently it has already been freed somewhere else (because the reference to that memory is a null pointer). Anyway, if you would checkout that branch and re-test your code and post you findings? That would be great!

Thanks.

StritziK, 2015-11-11:
Thanks for your help!
Now this is a little bit awkward: I Tested it more then once and getting different errors at different times. Maybe there are some problems with the combination of bluetooth-serial-port and the cronjob library or because I'm reusing the Bluetooth serial Object (I also tested it with instantiating a new Bluetooth module for each Job with the same effect)

* Atfer closing connection for the first time:
```console
(lldb) bt
* thread #8: tid = 0x1b2dfb, 0x00007fff8820448d libsystem_pthread.dylib`pthread_mutex_lock, stop reason = EXC_BAD_ACCESS (code=1, address=0x88)
  * frame #0: 0x00007fff8820448d libsystem_pthread.dylib`pthread_mutex_lock
    frame #1: 0x00000001023cd824 BluetoothSerialPort.node`pipe_consumer_free(handle=0x0000000000000000) + 36 at pipe.c:671
    frame #2: 0x00000001023cff2a BluetoothSerialPort.node`BTSerialPortBinding::EIO_Read(req=0x0000000101891808) + 234 at BTSerialPortBinding.mm:165
    frame #3: 0x000000010048c666 node`worker + 90
    frame #4: 0x000000010049756e node`uv__thread_start + 25
    frame #5: 0x00007fff882069b1 libsystem_pthread.dylib`_pthread_body + 131
    frame #6: 0x00007fff8820692e libsystem_pthread.dylib`_pthread_start + 168
    frame #7: 0x00007fff88204385 libsystem_pthread.dylib`thread_start + 13
```

* Atfer closing connection for the second time:
```console
* thread #8: tid = 0x1bfba3, 0x00007fff88460d5d libobjc.A.dylib`objc_msgSend + 29, stop reason = EXC_BAD_ACCESS (code=1, address=0x18)
  * frame #0: 0x00007fff88460d5d libobjc.A.dylib`objc_msgSend + 29
    frame #1: 0x00007fff9074045c CoreFoundation`-[NSDate laterDate:] + 60
    frame #2: 0x00000001024cc302 BluetoothSerialPort.node`-[BluetoothWorker getRFCOMMChannelIDTask:](self=0x0000000100c3a2c0, _cmd="getRFCOMMChannelIDTask:", address=@"00-19-5d-25-3f-5a") + 482 at BluetoothWorker.mm:315
    frame #3: 0x00007fff8108f41e Foundation`__NSThreadPerformPerform + 279
    frame #4: 0x00007fff906f98b1 CoreFoundation`__CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
    frame #5: 0x00007fff906d90ac CoreFoundation`__CFRunLoopDoSources0 + 556
    frame #6: 0x00007fff906d85cf CoreFoundation`__CFRunLoopRun + 927
    frame #7: 0x00007fff906d7fc8 CoreFoundation`CFRunLoopRunSpecific + 296
    frame #8: 0x00007fff8103cf59 Foundation`-[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 270
    frame #9: 0x00007fff8103ce38 Foundation`-[NSRunLoop(NSRunLoop) run] + 74
    frame #10: 0x00000001024cb40c BluetoothSerialPort.node`-[BluetoothWorker startBluetoothThread:](self=0x0000000100c3a2c0, _cmd="startBluetoothThread:", arg=0x0000000000000000) + 300 at BluetoothWorker.mm:120
    frame #11: 0x00007fff8104dee4 Foundation`__NSThread__start__ + 1351
    frame #12: 0x00007fff882069b1 libsystem_pthread.dylib`_pthread_body + 131
    frame #13: 0x00007fff8820692e libsystem_pthread.dylib`_pthread_start + 168
    frame #14: 0x00007fff88204385 libsystem_pthread.dylib`thread_start + 13
```
* Atfer closing connection for the second time:
```console
(lldb) bt
* thread #1: tid = 0x1c09e2, 0x00007fff8845ebe5 libobjc.A.dylib`realizeClass(objc_class*) + 51, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  * frame #0: 0x00007fff8845ebe5 libobjc.A.dylib`realizeClass(objc_class*) + 51
    frame #1: 0x00007fff8845ecba libobjc.A.dylib`realizeClass(objc_class*) + 264
    frame #2: 0x00007fff88461bff libobjc.A.dylib`lookUpImpOrForward + 104
    frame #3: 0x00007fff88460e11 libobjc.A.dylib`objc_msgSend + 209
    frame #4: 0x00007fff8103c89e Foundation`+[NSString stringWithCString:encoding:] + 139
    frame #5: 0x00000001023d1823 BluetoothSerialPort.node`BTSerialPortBinding::Close(info=0x00007fff5fbff550) + 259 at BTSerialPortBinding.mm:332
    frame #6: 0x00000001023c6f69 BluetoothSerialPort.node`Nan::imp::FunctionCallbackWrapper(info=0x00007fff5fbff5a8) + 153 at nan_callbacks_12_inl.h:174
    frame #7: 0x000000010013233c node`v8::internal::FunctionCallbackArguments::Call(void (*)(v8::FunctionCallbackInfo<v8::Value> const&)) + 160
    frame #8: 0x000000010014e786 node`v8::internal::Builtin_HandleApiCall(int, v8::internal::Object**, v8::internal::Isolate*) + 554
    frame #9: 0x00000e0458c0740e
    frame #10: 0x00000e0458f5e8cd
    frame #11: 0x00000e0458f5e722
    frame #12: 0x00000e0458ead937
    frame #13: 0x00000e0458c56fa0
    frame #14: 0x00000e0458c28891
    frame #15: 0x00000001001cf9da node`v8::internal::Invoke(bool, v8::internal::Handle<v8::internal::JSFunction>, v8::internal::Handle<v8::internal::Object>, int, v8::internal::Handle<v8::internal::Object>*) + 434
    frame #16: 0x00000001001206b1 node`v8::Function::Call(v8::Handle<v8::Value>, int, v8::Handle<v8::Value>*) + 213
    frame #17: 0x00000001004210e3 node`node::AsyncWrap::MakeCallback(v8::Handle<v8::Function>, int, v8::Handle<v8::Value>*) + 517
    frame #18: 0x000000010045770d node`node::TimerWrap::OnTimeout(uv_timer_s*) + 131
    frame #19: 0x0000000100497ef1 node`uv__run_timers + 38
    frame #20: 0x000000010048e4e3 node`uv_run + 580
    frame #21: 0x000000010042f0b6 node`node::Start(int, char**) + 344
    frame #22: 0x0000000100001b34 node`start + 52
```

eelcocramer, 2015-11-11:
Is this on the `issue87` branch?

eelcocramer, 2015-11-11:
Do the issues occur if you do not use cron jobs?

StritziK, 2015-11-11:
Yes I included the branch via 
```console
npm install https://github.com/eelcocramer/node-bluetooth-serial-port.git#issue87
```
I have rewritten my code to connect when calling a route without using cronjob. 
```javascript
var express = require('express'),
    BtConnector = require('../routes/btConnector').BtConnector,
    btConnector = new BtConnector(),
    router = express.Router();

router.route('/test')
    .post(function(req, res) {
        var btCommand = req.body;
        btConnector.exec(btCommand.command);
        res.send({
            message: 'message send'
        });
    });

module.exports = router;
```
I still get an error after connecting for a third time:

```console
(lldb) bt
* thread #8: tid = 0x1fa463, 0x00007fff88460d69 libobjc.A.dylib`objc_msgSend + 41, stop reason = EXC_BAD_ACCESS (code=1, address=0x1082ea000)
  * frame #0: 0x00007fff88460d69 libobjc.A.dylib`objc_msgSend + 41
    frame #1: 0x00007fff9067d56d CoreFoundation`-[__NSArrayM dealloc] + 205
    frame #2: 0x00007fff903a6755 IOBluetooth`-[IOBluetoothSDPDataElement dealloc] + 41
    frame #3: 0x00007fff906b2a0d CoreFoundation`-[__NSDictionaryI dealloc] + 189
    frame #4: 0x00007fff90398a18 IOBluetooth`-[IOBluetoothSDPServiceRecord dealloc] + 33
    frame #5: 0x00007fff90398a9b IOBluetooth`-[IOBluetoothSDPServiceRecord release] + 61
    frame #6: 0x00007fff906840c8 CoreFoundation`-[__NSArrayI dealloc] + 120
    frame #7: 0x00007fff88463ebb libobjc.A.dylib`(anonymous namespace)::AutoreleasePoolPage::pop(void*) + 477
    frame #8: 0x00007fff90697d02 CoreFoundation`_CFAutoreleasePoolPop + 50
    frame #9: 0x00007fff810bab0f Foundation`-[NSConnection handlePortCoder:] + 685
    frame #10: 0x00007fff810ba431 Foundation`-[NSConnection dispatchWithComponents:] + 53
    frame #11: 0x00007fff810b9ee9 Foundation`__NSFireMachPort + 254
    frame #12: 0x00007fff906e1242 CoreFoundation`__CFMachPortPerform + 290
    frame #13: 0x00007fff906e1109 CoreFoundation`__CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 41
    frame #14: 0x00007fff906e1079 CoreFoundation`__CFRunLoopDoSource1 + 473
    frame #15: 0x00007fff906d8aab CoreFoundation`__CFRunLoopRun + 2171
    frame #16: 0x00007fff906d7fc8 CoreFoundation`CFRunLoopRunSpecific + 296
    frame #17: 0x00007fff8103cf59 Foundation`-[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 270
    frame #18: 0x00007fff8103ce38 Foundation`-[NSRunLoop(NSRunLoop) run] + 74
    frame #19: 0x00007fff8104dee4 Foundation`__NSThread__start__ + 1351
    frame #20: 0x00007fff882069b1 libsystem_pthread.dylib`_pthread_body + 131
    frame #21: 0x00007fff8820692e libsystem_pthread.dylib`_pthread_start + 168
    frame #22: 0x00007fff88204385 libsystem_pthread.dylib`thread_start + 13
```

Maybe I will try it with a different machine

eelcocramer, 2015-11-12:
That last backtrace no classes from this module are mentioned. Weird. Do you get the same results on node v4?

I will retry your code again in an attempt to reproduce.

eelcocramer, 2015-11-17:
I've tested with the scripts you posted above and although cannot reproduce the error I see that stuff is going south when the `serial` object is reused several times.

I can get around that issue by creating a new instance on every iteration (so create a new `serial` object upon every request) which is probably a good thing to do in multithreading environment like a web server anyway.

Still the fact that the object is not re-usable on the long run seems to be a bug.

StritziK, 2015-11-23:
I have done some further testing and rebuilt it to always create an instance when connecting to a Bt-device. Still I get some error when running multiple times. This time there are also parts in bluetooth-serial-port.

```console
(lldb) bt
* thread #8: tid = 0x12f2de, 0x00007fff8cfbad57 libobjc.A.dylib`objc_msgSend + 23, stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
  * frame #0: 0x00007fff8cfbad57 libobjc.A.dylib`objc_msgSend + 23
    frame #1: 0x00007fff932e54c2 Foundation`+[NSProxy isAncestorOfObject:] + 25
    frame #2: 0x00007fff932e957c Foundation`-[NSKeyedPortCoder encodeObject:forKey:] + 67
    frame #3: 0x00007fff932e2da4 Foundation`-[NSConnection sendInvocation:internal:] + 987
    frame #4: 0x00007fff9f935472 CoreFoundation`___forwarding___ + 514
    frame #5: 0x00007fff9f9351e8 CoreFoundation`__forwarding_prep_0___ + 120
    frame #6: 0x00007fff9898bbe5 IOBluetooth`-[IOBluetoothDevice _performSDPQuery:uuids:] + 102
    frame #7: 0x0000000102c08210 BluetoothSerialPort.node`-[BluetoothWorker getRFCOMMChannelIDTask:](self=0x0000000100b07d50, _cmd="getRFCOMMChannelIDTask:", address=@"00-19-5d-25-3f-5a") + 240 at BluetoothWorker.mm:306
    frame #8: 0x00007fff932bc41e Foundation`__NSThreadPerformPerform + 279
    frame #9: 0x00007fff9f95a8b1 CoreFoundation`__CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
    frame #10: 0x00007fff9f93a0ac CoreFoundation`__CFRunLoopDoSources0 + 556
    frame #11: 0x00007fff9f9395cf CoreFoundation`__CFRunLoopRun + 927
    frame #12: 0x00007fff9f938fc8 CoreFoundation`CFRunLoopRunSpecific + 296
    frame #13: 0x00007fff93269f59 Foundation`-[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 270
    frame #14: 0x00007fff93269e38 Foundation`-[NSRunLoop(NSRunLoop) run] + 74
    frame #15: 0x0000000102c0740c BluetoothSerialPort.node`-[BluetoothWorker startBluetoothThread:](self=0x0000000100b07d50, _cmd="startBluetoothThread:", arg=0x0000000000000000) + 300 at BluetoothWorker.mm:120
    frame #16: 0x00007fff9327aee4 Foundation`__NSThread__start__ + 1351
    frame #17: 0x00007fffa06b89b1 libsystem_pthread.dylib`_pthread_body + 131
    frame #18: 0x00007fffa06b892e libsystem_pthread.dylib`_pthread_start + 168
    frame #19: 0x00007fffa06b6385 libsystem_pthread.dylib`thread_start + 13
```
I don't know if this is important but when I build it with 
```
node-gyp clean configure build --debug
```
there are following warnings
```console
../src/osx/BluetoothWorker.mm:117:39: warning: null passed to a callee that requires a non-null argument [-Wnonnull]
    keepAliveTimer = [[NSTimer alloc] initWithFireDate:[NSDate distantFuture]
                                      ^
../src/osx/BluetoothWorker.mm:117:39: warning: null passed to a callee that requires a non-null argument [-Wnonnull]
    keepAliveTimer = [[NSTimer alloc] initWithFireDate:[NSDate distantFuture]
```

eelcocramer, 2015-11-24:
Thanks for debugging this. I really appreciate it. The next 2 weeks I won't be able to spend any time on this but I'll dive into as soon as I have. :+1:

eelcocramer, 2019-08-23:
Not sure if this is still relevant. Closing for now.

closed 2019-08-23 by eelcocramer