eelcocramer/node-bluetooth-serial-port #88: Async inquire()?

jwynveen, 2015-12-07:
Is there any way to execute `btSerial.inquire();` asynchronously? When I execute that method, it locks my app for at least 10 seconds while it looks for devices, but I would prefer to show some loading icon/animation to show that work is happening so that users don't think the app is broken.

eelcocramer, 2015-12-08:
Currently not. I've tried to make this work in the past but ran into issues.

I have no plans on implementing this right now but I would be happy to accept a pull request that adds this feature. For OSX this should probably be done by putting the device inquiry in its own thread. The inquiry is started in the [worker](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/osx/BluetoothWorker.mm#L278).

As a workaround you might want to check if it is possible to fire up your [own thread from nodejs](http://stackoverflow.com/questions/18613023/how-to-create-threads-in-nodejs).

jwynveen, 2015-12-08:
I ended up using child_process.fork, as shown in http://jlunaquiroga.blogspot.com/2014/03/creating-processes-in-nodejs.html

eelcocramer, 2015-12-09:
Thank you for the feedback. I really appreciate it and I hope it will be helpful for others.

closed 2015-12-09 by eelcocramer

jonolomu, 2016-06-24:
I've had a go at trying to get this to work with [webworker-theads](https://github.com/audreyt/node-webworker-threads) but I am having issues.  I'm pretty new to node and javascript, so it may be obvious why it doesn't work, but the inquire() seems to never complete.

```javascript
var btSerial = new (require('bluetooth-serial-port')).BluetoothSerialPort();
var Worker = require('webworker-threads').Worker;

var worker = new Worker(function(){
  btSerial.inquire();
  self.close();
});
```

If I remove the inquire(), then the worker completes.

Any ideas? Is my best option to use the child_process.fork()?

Thanks!

eelcocramer, 2016-10-12:
Removing this as FAQ item as async inquiry will be added in the next release of this module. See also #95 and #118