eelcocramer/node-bluetooth-serial-port #89: is there any limit for receiving data size ?

rusyasoft, 2015-12-13:
Hi
 is there any limit for receiving data size ?

because i 'm not receiving sent message at once,  "serial.on('data', function(data)) { console.log('Received: ' + data);  });" gets triggered multiple times, where it supposed to get triggered once.

Output Example:
Received: f
Received: lame:NoFir
Received: e

actually sent message is "flame:NoFire\n"

evnrionment:
Arduino with BluetoothShield sends message to IntelEdison where i'm using your library with receiving code.

p.s: By the way Thank You a lot, this library is really good !

Rustam

eelcocramer, 2015-12-13:
Thanks!

Please see the FAQ #10 for this. As the module is unaware of your datamodel (ie a message ends with a `\n` it just reacts to events from the Bluetooth layer that there is incoming data and passes that on to your application, you should buffer the incoming data and recreate the message from the buffer.

Good luck!

closed 2015-12-13 by eelcocramer

rusyasoft, 2015-12-13:
Yeah thats what I did :)

Thanks a lot