eelcocramer/node-bluetooth-serial-port #90: Close() or Disconnection doesn't work

rusyasoft, 2016-01-03:
Hi there

Close() function doesn't work?
What can i do ?

the code is quite simple.

Environment.
From IntelEdison i'm connecting to HC06 bluetooth device on arduino. 
Connection works fine, but when i'm calling for disconnection it doesn't work. I mean the Bluetooth shield doesn't start blinking (blinking means no connection). So the led stays still which refers to the existence of connection between edison and arduino.

Could you tell me what is that ?

Thanks in advance
Rustam

rusyasoft, 2016-01-03:
Now it seems to be disconnecting well but it gives an error after connecting second time

"Error reading from connection" and then NaN objects 

oh right, i'm getting every message two times. Which means even though i already read the buffer, next time i read it i read again same message. It feels like there are two BluetoothSerial objects :(

when the disconnection finished should i delete the BluetoothSerial object ?

rusyasoft, 2016-01-03:
Ok, got it I found it.

Conclusion in Short: Recreate the BluetoothSerial Object next time you wanna reconnect to your bluetooth Device

Details:
After closing the BluetoothSerial object we better delete it once. So next time we need it we should recreate it again. Otherwise it keeps inside some previous connections or buffers, i dont know but something weird happens.

@eelcocramer is this problem familiar to you ?

eelcocramer, 2016-01-03:
Thanks for the report and posting your solution. This is all on Linux right?

Does not sound familiar to me but reusing the object was never a use case for me. When you reuse the object and create a new connection do you register a callback again? Closing the connection does not clear the registered callbacks I think, which for some people might be desired behaviour.

rusyasoft, 2016-01-03:
Yes its all Linux (Yocto linux on Intel Edison)

I think now it makes sense. Because I have registered callbacks for 'close' on top level. But 'data' callback was registered inside the 'connect' event. Thats why I had two 'data' event triggering. I will try to declare them all on top level and try again. But so far delete and recreation works fine :)

Thank You !

closed 2016-04-28 by eelcocramer