eelcocramer/node-bluetooth-serial-port #91: (Question) How to compile for nwjs (mac)

EdisonDelgado, 2016-01-21:
Hello,
I get this error when i try to use in node-webkit v.0.12.3 and OSX 10.11.1
```Module version mismatch. Expected 43, got 46.```

What is the correct way to compile for nwjs?
try```node-gyp configure build```compiled, but nothing happened and don't error generate
I have installed  
node v4.2.1
npm v2.14.7
node-gyp v3.2.1

[image]

thank in advance!

eelcocramer, 2016-01-22:
Did you check the solution of #73?

ELundby45, 2016-03-03:
This is what is working for me:

You need to use [nw-gyp](https://github.com/nwjs/nw-gyp) instead of node-gyp. 

```
#Install nw-gyp
npm install -g nw-gyp

#change the directory with bluetooth-serial-pot
cd node_modules/bluetooth-serial-port

#Build with nw-gyp be sure to replace $NWJS_VERSION with your actual version ie: 0.8.6
nw-gyp clean
nw-gyp configure --target=$NWJS_VERSION --arch=ia32
nw-gyp build
```

eelcocramer, 2016-03-03:
Thanks for posting!

closed 2016-04-28 by eelcocramer