eelcocramer/node-bluetooth-serial-port #92: Bluetooth server

alexnix, 2016-02-04:
Hi,

I understand this module can be used to discover and connect to devices, but is it capable to create a server for other devices to discover and connect to? Actually is there any node module for this? I could only find Java and Python examples...

Thanks

eelcocramer, 2016-04-28:
Hi, sorry for the late response. This module is not capable to act as a server. I'm not aware of any other node modules that have this feature.

atilag, 2016-08-09:
Hi @eelcocramer @alexnix , I have made the server support for RFCOMM sockets on Linux.
Some of the features are:
* Asynchronous listening for incoming RFCOMM connections 
* Register the service via SDP so it gets advertised
* Backward compatible with client-code (barely haven't touched client-code).
Only thing left... update the doc. :)
Instead of doing a complete new module, I thing that having just one Bluetooth "Classic" module for everything is way more clear and easy to use/maintain for everybody. @eelcocramer  I'm gonna PR so you can take a look and see if there's something would like to change.
Can take a look to experiments/server-test.js to see how to use it.
Regards!

eelcocramer, 2016-08-09:
Very cool! I will look at it when I'm back home in the end of August.

eelcocramer, 2016-09-13:
Thanks to @atilag this is available now in version 1.3.1 that was just published to npm.

closed 2016-09-13 by eelcocramer