eelcocramer/node-bluetooth-serial-port #93: Putting code inside of a module fails to find serial port

MrZunz, 2016-02-28:
Working code:

```
// var obd = require('obd');
// obd.connect();

var BluetoothSerialPort = require("bluetooth-serial-port").BluetoothSerialPort;
var myBtPort = new BluetoothSerialPort();
var dataBuffer = "";

myBtPort.on('found', function (address, name) {
    console.log('Found: ' + address + ' with name ' + name);

    if(name != "ubuntu-0" && name != "OBDII") { return; }

    myBtPort.findSerialPortChannel(address, function(channel) {
        console.log('Found RFCOMM channel for serial port on ' + name + ': ' + channel);
        console.log('Attempting to connect...');

        myBtPort.connect(address, channel, function() {
            console.log('Connected. Receiving data...');
            myBtPort.on('data', function(buffer) {

            });
        });
    });
});

myBtPort.on('finished', function() {
    console.log('scan did finish');
});

myBtPort.inquire();
```

This does work every time I run it, it'll find the serial port and connect just fine. However, when I put the exact same code inside of a module it doesn't. It wont find the serial port...

```
function OBD() {
	console.log("OBD:", 'Initializing...');
}

OBD.prototype.connect = function() {

	var BluetoothSerialPort = require("bluetooth-serial-port").BluetoothSerialPort;
	var myBtPort = new BluetoothSerialPort();
	var dataBuffer = "";

	myBtPort.on('found', function (address, name) {
	    console.log('Found: ' + address + ' with name ' + name);

	    if(name != "ubuntu-0" && name != "OBDII") { return; }

	    myBtPort.findSerialPortChannel(address, function(channel) {
	        console.log('Found RFCOMM channel for serial port on ' + name + ': ' + channel);
	        console.log('Attempting to connect...');

	        myBtPort.connect(address, channel, function() {
	            console.log('Connected. Receiving data...');
	            myBtPort.on('data', function(buffer) {

	            });
	        });
	    });
	});

	myBtPort.on('finished', function() {
	    console.log('scan did finish');
	});

	myBtPort.inquire();
}

module.exports = exports = new OBD();
```
Anyone has any clue why it is behaving like this? This does not seem logical but I just can't get it to work with a module...

eelcocramer, 2016-02-28:
What is the console output for the 2nd code piece? I get:

```
OBD: Initializing...
scan did finish
```

for this program:

```javascript
var odb = require('./odb.js');
odb.connect();
```

But I do not have a test device that I can connect to right now.

MrZunz, 2016-02-28:
Output of the working code is

```
Found: C4:8E:8F:F9:B7:88 with name ubuntu-0
Found: 00:04:20:E9:05:28 with name Harmony Keyboard
scan did finish
Found RFCOMM channel for serial port on ubuntu-0: 1
Attempting to connect...
Connected. Receiving data...
```

And the not working code results in

```
OBD: Initializing...
Found: 00:04:20:E9:05:28 with name Harmony Keyboard
Found: C4:8E:8F:F9:B7:88 with name ubuntu-0
scan did finish
```

eelcocramer, 2016-02-28:
Sanity check. Does it break out in this line?

```
if(name != "ubuntu-0" && name != "OBDII") { return; }
```

Can you put `console.log`'s before and after this line?

MrZunz, 2016-02-28:
I added `console.log('Trying to find serial port on', name);` after the check and this is the output:
```
OBD: Initializing...
Found: 00:04:20:E9:05:28 with name Harmony Keyboard
Found: C4:8E:8F:F9:B7:88 with name ubuntu-0
Trying to find serial port on ubuntu-0
scan did finish
```

eelcocramer, 2016-02-28:
You can also try using different nodejs versions (0.12, 4 and 5) to see if that might cause something weird. Be sure to re-install / re-build the module when you try with a another version... You can easily switch between version using something like [nvm](https://github.com/creationix/nvm).

MrZunz, 2016-02-28:
I tried with node 4.0.0 and 4.3.1. Same behavior sadly, I'll see if I can try with another version!

eelcocramer, 2016-02-28:
Thanks, please let me know if you find anything. I will test myself when I get my device.

MrZunz, 2016-02-28:
I tried with 0.12.10, 5.0.0 and 5.7.0 and they all behave the same. I don't think it's nodejs version specific... :/

MrZunz, 2016-02-29:
Might also be note-worthy that I am running this code on a Raspberry Pi and trying to connect to my laptop running ubuntu.

eelcocramer, 2016-02-29:
Thank you for getting back with the test results. Are you able to test if it does work the other way around? It might be worth to rule out if this is a platform specific issue.

MrZunz, 2016-03-02:
I got my Raspberry Pi 3 yesterday and I am using nodejs 4.0 again and it works just fine now. Didn't get any further with this issue which was on a Raspberry Pi 2. So I guess you can close this issue for now! Thanks for all the help so far!

eelcocramer, 2016-03-02:
Ok thanks for letting me know. I will close and add this issue to the FAQ for future reference.

closed 2016-03-02 by eelcocramer