eelcocramer/node-bluetooth-serial-port #94: Cannot install the module on windows 10

shikhar-chaayos, 2016-03-07:
Environment: 
Visual C++ 2015
Python 3.5

When installing the module, I am seeing this issue:

npm ERR! bluetooth-serial-port@1.2.6 install: `node-gyp configure build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bluetooth-serial-port@1.2.6 install script 'node-gyp configure build'.
npm ERR! This is most likely a problem with the bluetooth-serial-port package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp configure build
npm ERR! You can get their info via:
npm ERR!     npm owner ls bluetooth-serial-port
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\shikhar\npm-debug.log

eelcocramer, 2016-03-14:
Thanks for your report.

There are a couple of [FAQ](https://github.com/eelcocramer/node-bluetooth-serial-port/issues?utf8=✓&q=label%3AFAQ+) issues about compiling the module on Windows. Do any of these help you out?

It would help if you would post the actual error you are getting (the stuff in the log before `Exit status 1`).

eelcocramer, 2016-04-28:
Is this still relevant?

closed 2016-04-28 by eelcocramer