eelcocramer/node-bluetooth-serial-port #95: Inquire function asynchronous

kncs, 2016-04-28:
It looks like that inquire function is synchronous, is there any way to render the inquire function asynchronous ?

eelcocramer, 2016-04-28:
Not currently. This would require changes in the native code (for all supported platforms).

I have no plans to change the current behaviour but I'm happy to accept a PR that makes the inquire function asynchronously.

kncs, 2016-04-28:
Ok, I'am working on it and did implement inquire as a asynchrnous function, I also did create a inquireSynch function. For the moment it's only working for linux, but I will implement it on all platforms asap in order to make a PR.

https://github.com/kncs/node-bluetooth-serial-port/tree/asyncInquire

eelcocramer, 2016-04-29:
Great work. You also might want to check out issue #88 for some previous work on this.

ollym, 2016-09-23:
@kncs any progress with this?

kncs, 2016-09-24:
@ollym , I did implement the solution for linux and windows, I never take the time to do it for mac. I should set a VM with a mac OS env to finish the job. If you are not a mac user, you can try it using my fork.

ollym, 2016-09-25:
@kncs why not just do a pull request the linux / windows? Then someone with an OSX environment could implement that separately.

kncs, 2016-10-10:
@ollym, sorry for the delay. Last week I did set a VM with the last mac OS X Sierra but I was not able to attach my broadcom bluetooth adaptater to the VM... You certainly right, someone with an osx env could implement that separately.

eelcocramer, 2016-10-12:
The OSX implementation of this feature has been added. Cross platform support for this feature is now included in the `next-release` branch which will be published somewhere in the upcoming days.

eelcocramer, 2016-10-13:
Version 2.1.0 that adds this feature was released today.

closed 2016-10-13 by eelcocramer