eelcocramer/node-bluetooth-serial-port #96: EventEmitter memory leak

ColdFireIce, 2016-04-29:
Hi,

after closing and opening the BT connection several times i get this warning:

```
(node) warning: possible EventEmitter memory leak detected. 11 data listeners added. Use emitter.setMaxListeners() to increase limit.
project Trace
at BluetoothSerialPort.addListener (events.js:239:17)
at /opt/project/node_modules/bluetooth-serial-port/lib/bluetooth-serial-port.js:87:22
```

I made sure to close() the btSerial object. isOpen() reported false.
I also get an increasing number of 'data' events when calling the btSerial.close() function. The returned buffer is always empty for these events.

Greetings

eelcocramer, 2016-04-29:
Hi, thanks for your report. Can you post your code? What OS are you using?

You should be aware that `close` does not clear the emitters. So when you re-open the BT connection you shouldn't have to add new listeners.

ColdFireIce, 2016-04-29:
Hi,

no I'm not adding new event listener. OS is Raspbian (Debian). I'll try to create a code snippet next week, which reproduces the error.

Greetings

eelcocramer, 2016-09-13:
Closing this for now.

closed 2016-09-13 by eelcocramer