eelcocramer/node-bluetooth-serial-port #97: Can't exit proccess

ColdFireIce, 2016-04-29:
Hi,

It seems that after calling connect() and close() a couple times, some background tasks prevent the process from closing.

After calling process.exit() the process.on('exit') event is emitted, but the process stays open. The only way to kill the program is to use the kill -9 PID shell command from outside. (Ctrl+X, Ctrl+C won't work either)

Greetings

eelcocramer, 2016-04-29:
Thank you for the report. What OS are you on? Can you post code to reproduce this?

ColdFireIce, 2016-04-29:
Hi,

OS is Raspbian (Debian). I'll try to create a code snippet next week, which reproduces the error.

Greetings

eelcocramer, 2016-09-13:
Closing this for now.

PsychoXIVI, 2019-08-18:
Experiencing the similar issue :/

PsychoXIVI, 2019-08-18:
I found `lsof -p <PID>` useful to check which FDs could make node to wait...
```
COMMAND  PID USER   FD      TYPE DEVICE SIZE/OFF            NODE NAME
node    2256   pi  cwd       DIR   0,33     4096 562949953446264 /mnt/earth/berry
node    2256   pi  rtd       DIR  179,2     4096               2 /
node    2256   pi  txt       REG  179,2 38740596            1711 /usr/local/lib/nodejs/node-v12.8.1-linux-armv7l/bin/node
node    2256   pi  mem       REG   0,33                187038577 /mnt/earth/berry/node_modules/bluetooth-serial-port/build/Release/BluetoothSerialPortServer.node (path inode=5629499534238240)
node    2256   pi  mem       REG  179,2   132056            1831 /usr/lib/arm-linux-gnueabihf/libbluetooth.so.3.18.14
node    2256   pi  mem       REG   0,33               1687341369 /mnt/earth/berry/node_modules/bluetooth-serial-port/build/Release/BluetoothSerialPort.node (path inode=6192449487659536)      
node    2256   pi  mem       REG  179,2  1234700            8045 /lib/arm-linux-gnueabihf/libc-2.24.so
node    2256   pi  mem       REG  179,2   127300            8059 /lib/arm-linux-gnueabihf/libpthread-2.24.so
node    2256   pi  mem       REG  179,2   116372            1878 /lib/arm-linux-gnueabihf/libgcc_s.so.1
node    2256   pi  mem       REG  179,2   452152            8049 /lib/arm-linux-gnueabihf/libm-2.24.so
node    2256   pi  mem       REG  179,2  1269620           10573 /usr/lib/arm-linux-gnueabihf/libstdc++.so.6.0.22
node    2256   pi  mem       REG  179,2    26632            8061 /lib/arm-linux-gnueabihf/librt-2.24.so
node    2256   pi  mem       REG  179,2     9800            8048 /lib/arm-linux-gnueabihf/libdl-2.24.so
node    2256   pi  mem       REG  179,2    21868            5647 /usr/lib/arm-linux-gnueabihf/libarmmem.so
node    2256   pi  mem       REG  179,2   138576            8040 /lib/arm-linux-gnueabihf/ld-2.24.so
node    2256   pi    0u      CHR  136,0      0t0               3 /dev/pts/0
node    2256   pi    1u      CHR  136,0      0t0               3 /dev/pts/0
node    2256   pi    2u      CHR  136,0      0t0               3 /dev/pts/0
node    2256   pi   13u  a_inode   0,11        0            5804 [eventpoll]
node    2256   pi   14r     FIFO   0,10      0t0           16295 pipe
node    2256   pi   15w     FIFO   0,10      0t0           16295 pipe
node    2256   pi   16u  a_inode   0,11        0            5804 [eventfd]
node    2256   pi   17r     FIFO   0,10      0t0           16298 pipe
node    2256   pi   18w     FIFO   0,10      0t0           16298 pipe
node    2256   pi   19u     sock    0,8      0t0           16459 protocol: RFCOMM
node    2256   pi   21u      CHR  136,0      0t0               3 /dev/pts/0
node    2256   pi   22r      CHR    1,3      0t0            1028 /dev/null
```

eelcocramer, 2019-08-18:
> Experiencing the similar issue :/
> 
> ```js
> const wtfnode = require('wtfnode');
> wtfnode.init();
> const { BluetoothSerialPortServer } = require('bluetooth-serial-port');
> 
> var channel = 1;
> var UUID = '00001101-0000-1000-8000-00805F9B34FB';
> 
> var server = new BluetoothSerialPortServer();
> 
> server.on('data', (buffer) => {
>     console.log('< ' + buffer);
> });
> 
> server.listen((clientAddress) => {
>     console.log('Client: ' + clientAddress + ' connected!');
> }, (error) => {
>     console.error('Failed on listening: ' + error);
> }, {uuid: UUID, channel: channel} );
> 
> // Testing propouses:
> setTimeout(() => {
>     input.close();
>     server.disconnectClient();
>     server.close();
>     process.on('exit', () => {
>         console.log('exiting')
>         wtfnode.dump();
>     })
>     process.exit();
> }, 1000);
> ```
> 
> After `server.close()` node still stays open. Adding `process.exit()` does not help. There is `wtfnode` added to check what is blocking the node from exiting:
> 
> ```
> > node main.js
> 
> exiting
> [WTF Node?] open handles:
> - File descriptors: (note: stdio always exists)
>   - fd 1 (tty) (stdio)
> - Timers:
>   - (1000 ~ 1000 ms) (anonymous) @ /mnt/earth/berry/main.js:39
> ```
> 
> It seems like closing server does not free all file descriptors.
> 
> @eelcocramer, please reopen.

I'm not familiar with WTF but from the output above I conclude that only `stdio` is still open? Which is fine as it should alway be open?

PsychoXIVI, 2019-08-18:
It is not listing all the things - I guess because of the RFCOMM socket is used by binary module.

PsychoXIVI, 2019-08-18:
Okay, I found the issue, and successfully made it work :) I will prepare PR soon.

PR #222 (2019-08-21): Releases version 2.2.5
Resolves issue #97 and incorporates changes from PR #221 

Closes #97
Closes #221
closed 2019-08-21 by eelcocramer