eelcocramer/node-bluetooth-serial-port #98: Close doesn't seem to close the connection

opensourcegeek, 2016-05-28:
I've tried to close the connection after connect/write, but according to the device I'm connecting to, it seems like the connection is still open to the device. Unless I stop the process from running the devices look like they are connected to my laptop.

OS Fedora 22 (Linux), please let me know if you require any specific info. Please let me know if you require any further info.

opensourcegeek, 2016-05-31:
I cannot open another connection to the device after I've closed connection, however on stopping the process which spawned the connection I can reopen connection. As you've mentioned this could be a behaviour of my bluetooth driver/device itself too, but not sure how to establish where the issue lies. Any ideas? 

So currently I'm working around this issue by having a child process to do the comms and shutting the child process down after I'm done.

eelcocramer, 2016-05-31:
It is not possible to re-open a closed connection. However you should be able to create a new `BluetoothSerialPort` object and use that to re-connect to your device.

eelcocramer, 2016-05-31:
See also FAQ issue #90

gengshenghong, 2016-07-22:
@eelcocramer This thread is help for me to identify an bug in my test code (it failed when re-connect), but during some more testing, I find very strange behavior below, could you please help to take a look and explain why my code does not work when I "close" before "connect" in one function.

In below code, you can find below line:
// btDisconnect();      // with this line, it work!!!
With this line, the code will connect and reconnect the same device successfully. But without it, it failed in 2nd connection (the error is "[Error: Cannot connect]"). In below code, I just want the "btConnect" function to disconnect before connect automatically, but it cannot work. I cannot understand where it is blocking me to release the connection. Thank you very much.

`var BluetoothSerialPort = require('bluetooth-serial-port').BluetoothSerialPort;
var btSerial = null;
var isConnected = false;

// connect to bluetooth device as serial port
var btConnect = function(addr, callback) {
    if(isConnected) {   // if connected, disconnect it first
        btDisconnect();
    }

    setTimeout(function() {
        _btConnect(addr, callback);
    }, 1000);  // wait for some more time....
};

var _btConnect = function(addr, callback) {
    btSerial = new BluetoothSerialPort();

    btSerial.connect(addr, 1, function() {
        isConnected = true;
        callback(true);
    }, function (error) {
        console.log(error);
        callback(false);
    });
};

// disconnect the bluetooth serial port
var btDisconnect = function() {
    btSerial.close();
    isConnected = false;
};

var address = '00:BA:55:57:5F:67';

var count = 0;
var reconnectTest = function() {
    console.log("connecting...");
    btConnect(address, function(ret) {
        if (ret === true) {
            count = count + 1;
            console.log("connected...");

            setTimeout(reconnectTest, 100);
            // btDisconnect();      // with this line, it work!!!
        } else {
            console.log(count);
            console.log("not connected...");
            process.exit();
        }
    });
};

reconnectTest();`

gengshenghong, 2016-07-23:
More information, without the "btDisconnect" line, if I call "reconnectTest()" it works, using setTimeout or setImmidiate does not work, see below:

reconnectTest();      // works
           // setTimeout(reconnectTest, 100);  // not work!
           setImmediate(reconnectTest); // not work

The code is complete, I cannot understand what are the differences there.

eelcocramer, 2016-07-23:
Just to make things more clear to me:

**Without** calling `btDisconnect` like in your example above your code does work? But when you do call that function your code does not work?

So the problem is that you are able to *reconnect* if your connection has not been terminated but not when the connection has actually been closed?

gengshenghong, 2016-07-23:
@eelcocramer  In fact, it is called as below:
var btConnect = function(addr, callback) {
    if(isConnected) {   // if connected, disconnect it first
        btDisconnect();
    }

As mentioned, I just want to have a very simple wrapper "btConnect" which will disconnect before connect. Logically, the btDisconnect is not needed, I've struggled for a long time to analysis the logic and I do not see any issue in the logic. Maybe it is related to some other issues in low-level stuff in nodejs, which I am not quite familiar with. :(

gengshenghong, 2016-07-23:
I've tested the code on both Ubuntu 16.04 and 14.04, using node v5.12.0.
Also, I mentioned "without the btDisconnect line, it failed in 2nd connection", in fact, sometimes, it may be successful for more times (like 10 times, 30 times...which is the value of "count" in my code), but with that line, the re-connection can be successful always (I've tested for 30 minutes at least).

opensourcegeek, 2016-07-23:
@gengshenghong It's interesting you're seeing a similar sort of issue (if I understand correctly) - one side note is when you call close it doesn't close connection to your device immediately though, you'd have to wait for `'closed'` event to come through. Example code below (uses async/await, I find it as a much better approach if you're doing serial comms in nodejs rather than using callbacks.)

```
export function disconnectDevice(btSerial, deviceName) {
    return new Promise((resolve, reject) => {
        btSerial.on('closed', () => {
            console.log('Device disconnected ', deviceName);
            resolve();
        });
        btSerial.close();
    });
}

await disconnectDevice(btSerial, device.name);
```
However even after I get 'closed' event fired, my laptop still had connection open to the device. Only way I got around that problem was by having a child process do all the comms work and shutting it down once I'm done. So when I had to start comms again with same device there's no connection hangovers from previous time.

gengshenghong, 2016-07-23:
@opensourcegeek Yes, I am also considering using child process to make sure everything is stable, but that may have a little performance impact (we use bluetooth for short-message communication usually, example send  control command to an Arduino board), with child process, all the communication need to send to child process, I've not tested but I am also considering using child process if there is no other choices. :(
Technically, it seems like there are something wrong and may be fixed, hope @eelcocramer have more idea on it. In my test code, putting "close()" in different places get different result, but logically, they should behavior same.

Regarding the "closed" event, I'm aware of it. That is why I've used the "setTimeout" in my code to wait for a while (I've waited for 1 seconds and I've confirmed "closed" event is fired, I've even tried to wait for 5 seconds or longer, but still having issues). The above code are just for test purpose, in real code, we'll need to wait for "closed" event instead of using "setTimeout".
`setTimeout(function() {
        _btConnect(addr, callback);
    }, 1000);  // wait for some more time....`

eelcocramer, 2016-07-25:
You could attempt to debug the close function in the Linux C code. I has been 9 months since I last touched this code and I must admit that scanning through the code it makes my eyebrows frown. You might want to debug the flow of the close function to see what actually happens. For example it is unclear to me what are in these file descriptors [1](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/linux/BTSerialPortBinding.cc#L353), [2](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/linux/BTSerialPortBinding.cc#L359) and why `close` is being written to one of them (this might be code ported from the OSX implementation).

The actual closing of the connection happens [here](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/linux/BTSerialPortBinding.cc#L352). There is not much that can go wrong there.

Do you have the same issue with node v4?

gengshenghong, 2016-07-27:
I've rewrite my code using child_process, which is very stable. The same issue happens with using default official node shipped in Ubuntu 14.04 (v0.10.37), it seems like not an issue of node itself.

closed 2016-10-12 by eelcocramer