eelcocramer/node-bluetooth-serial-port #99: Unable to install on Windows 10 + Visual Studio 2015

barchard, 2016-06-07:
Hi,

When running command `npm install bluetooth-serial-port --msvs_version=2015`

### Python
```
>python --version
2.7.11
```

### Visual Studio
```
>dir "%ProgramFiles(x86)%\Microsoft SDKs\Windows"
 Volume in drive C is TI10707300B
 Volume Serial Number is B8D2-2E2B

 Directory of C:\Program Files (x86)\Microsoft SDKs\Windows

06/06/2016  02:00 PM    <DIR>          .
06/06/2016  02:00 PM    <DIR>          ..
06/06/2016  01:33 PM    <DIR>          v10.0A
06/06/2016  01:34 PM    <DIR>          v7.0A
06/06/2016  02:00 PM    <DIR>          v8.1
06/06/2016  01:54 PM    <DIR>          v8.1A
               0 File(s)              0 bytes
               6 Dir(s)  743,683,301,376 bytes free
```

### Error
```
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Platforms\x64\PlatformToolsets\v140\Toolset.targets(36,5): error MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK or change the SDK version in the project property pages or 
```

eelcocramer, 2016-06-07:
So I do not have a windows environment so I'm probably not the best person to help you 😅 

I'm not sure if visual studio 2015 is supported. Do you have 2012? Can you try: `npm install bluetooth-serial-port --msvs_version=2012`

provstevi, 2016-06-14:
I was able to install using VS 2015 once I made sure I'd actually installed all the C++ tools. But I am unable to do a nw-gyp build due to the following exception: 

..\src\windows\BTSerialPortBinding.cc(39): warning C4996: 'WSAStringToAddressA': Use WSAStringToAddressW() instead or d
efine _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings [C:\Users\Providence\node_modules\bluetooth-se
rial-port\build\BluetoothSerialPort.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(3623): note: see declaration of 'WSAStringToAddressA'
..\src\windows\BTSerialPortBinding.cc(192): error C2660: 'node::FatalException': function does not take 1 arguments [[node dir]\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]

eelcocramer, 2016-06-15:
What version of node are you using? Can you provide the output of `npm ls`?

Thanks.

eelcocramer, 2016-09-13:
Closing this for now.

closed 2016-09-13 by eelcocramer