eelcocramer/node-bluetooth-serial-port #100: Fatal error after sending one message

Pravez, 2016-06-14:
Hey ! I am trying to use your library to send message to multiple robots connected with Bluetooth and using serial ports, at the same time. Problem is that when I am connected, and also I have sent a message, I get a fatal error concerning V8, here is the error message ... :

#
# Fatal error in , line 0
# unreachable code
#

==== C stack trace ===============================

 1: V8_Fatal
 2: 0x7fb36c501547
 3: v8::internal::ToBooleanICStub::UpdateStatus(v8::internal::Handle<v8::internal::Object>)
 4: v8::internal::Runtime_ToBooleanIC_Miss(int, v8::internal::Object**, v8::internal::Isolate*)
 5: 0x680c8e092a7

Have you an idea about this ...?

eelcocramer, 2016-06-14:
Hi,

Thanks for your message.

Can you add more details (node version etc) and provide some sample code that can be used to reproduce the error?

Cheers!

Pravez, 2016-06-14:
Ok finally after some researches, problem solved. It comes from the electron version. Indeed, <quote> serialport becomes a nightmare with electron 1.2.0</quote> (from here : [Issue serial port](https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/838)).
So I am just going back to v1.1.2 to have it working correctly.

Thank you ... :3

eelcocramer, 2016-06-14:
Thank you for the report. Good luck! 👍

closed 2016-06-14 by eelcocramer