eelcocramer/node-bluetooth-serial-port #101: node v6 deprecation warning

dcornejo, 2016-06-14:
I am getting this error when using this package with node v6.2.1

```
(node) v8::ObjectTemplate::Set() with non-primitive values is deprecated
(node) and will stop working in the next major release.
```
(full traces attached)
[session.txt](https://github.com/eelcocramer/node-bluetooth-serial-port/files/315082/session.txt)

there are several other modules where this has cropped up and discussion about how they resolved it, I can provide links to those if needed.

eelcocramer, 2016-06-15:
What is the output of `npm ls`? I suspect that you have a old version of `nan` lingering around.

dcornejo, 2016-06-15:
you are correct, sorry!

closed 2016-06-15 by dcornejo

eelcocramer, 2016-06-16:
No problem. Thank your for providing feedback.