eelcocramer/node-bluetooth-serial-port #102: Intel Edison Bluetooth

pratikus, 2016-07-19:
Hello Eelco,

We are kind of stuck at a critical point in the project and are hoping you could point us in the right direction.

We are trying to get two Intel Edisons to talk to each other/cellphone via Bluetooth.

Here is an image of our test-setup: http://imgur.com/4LXQtna
[image]

Here is our main program: http://imgur.com/a/vaZoB
[image]
[image]

Here is a test.js output: http://imgur.com/EJMjzla
[image]

So far we have only been able to connect to an iphone via Bluetooth (please see main program link). We are not able to connect the Edison’s to each other. As such, we don’t know if we can send data back and forth with your module.

Really hoping you can get involved in this project!

eelcocramer, 2016-07-20:
Hi. Looks like a nice project! I've received your mail via linkedin as well but prefer to answer this here in the true spirit of open source 👍🏼

Have you checked the [FAQ](https://github.com/eelcocramer/node-bluetooth-serial-port/issues?utf8=✓&q=label%3AFAQ%20)?

FAQ entry #69 might be relevant for your: this library is client side only. If you have BT serial port service running somewhere you can use this library to connect to it. I'm unaware of any node modules that implement the server side.

I hope this helps.

noahko123, 2016-07-20:
Hi Eelco,

Would the github project for noble BLE (https://github.com/sandeepmistry/noble) work as the server for your project?

eelcocramer, 2016-07-20:
No, that is a Bluetooth Low Energy implementation where this client implements the Bluetooth serial port profile. These are completely different Bluetooth profiles.

noahko123, 2016-07-20:
Thanks for the quick response.  You've been a big help!

eelcocramer, 2016-07-21:
No problem I'm glad I could help

closed 2016-07-21 by eelcocramer