eelcocramer/node-bluetooth-serial-port #106: Unable to detect my device

spl0i7, 2016-09-12:
Hi,
It looks like I am unable to detect my mobile's bluetooth using this library. I used following code
```js
var btsp = new (require('bluetooth-serial-port')).BluetoothSerialPort();
btsp.on('found',(address,name)=>{
  process.stdout.write(`ADDRESS : ${address}\n`);
  process.stdout.write(`NAME : ${name}\n`);
});
btsp.inquire();
```
I can see my device using **btmon** as
```
 HCI Event: Extended Inquiry Result (0x2f) plen 255                                                                                                                         [hci0] 31.003253
        Num responses: 1
        Address: 00:00:00:00:5A:AD (XEROX CORPORATION)
        Page scan repetition mode: R1 (0x01)
        Page period mode: P0 (0x00)
        Class: 0x5a020c
          Major class: Phone (cellular, cordless, payphone, modem)
          Minor class: Smart phone
          Networking (LAN, Ad hoc)
          Capturing (Scanner, Microphone)
          Object Transfer (v-Inbox, v-Folder)
          Telephony (Cordless telephony, Modem, Headset)
        Clock offset: 0x1e78
        RSSI: -48 dBm (0xd0)
        Name (complete): Motorola Moto E
        16-bit Service UUIDs (complete): 11 entries
          OBEX Object Push (0x1105)
          Audio Source (0x110a)
          A/V Remote Control Target (0x110c)
          A/V Remote Control (0x110e)
          Headset AG (0x1112)
          PANU (0x1115)
          NAP (0x1116)
          Handsfree Audio Gateway (0x111f)
          Phonebook Access Server (0x112f)
          PnP Information (0x1200)
          Message Access Server (0x1132)
@ Device Found: 00:00:00:00:5A:AD (0) rssi -48 flags 0x0000
        10 09 4d 6f 74 6f 72 6f 6c 61 20 4d 6f 74 6f 20  ..Motorola Moto 
        45 17 03 05 11 0a 11 0c 11 0e 11 12 11 15 11 16  E...............
        11 1f 11 2f 11 00 12 32 11 01 05 01 07 04 0d 0c  .../...2........
        02 5a                                            .Z              

```
but somehow that code doesn't detects this device, although I can see other devices. I am using fedora with kernel version **4.6.7-300.fc24.x86_64** and my bluetooth device is Moto E. How can I fix this issue?

eelcocramer, 2016-09-12:
Does you phone support the bluetooth serial port profile? It is not listed in the services of your btmon log above.

This module only returns devices with that profile available.

spl0i7, 2016-09-12:
@eelcocramer that might explain this behavior, is there any workaround possible?

eelcocramer, 2016-09-12:
I'm afraid not. The service profile that the Bluetooth inquiry is looking for is hardcoded in the source code. If you want to search for another profile you need to change the sources. See [this line](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/linux/DeviceINQ.cc#L70) for the Linux code. Please note that this module only implemented the serial port profile and you cannot use it to connect to other Bluetooth profiles.

eelcocramer, 2016-10-12:
Closing this because the module shows it intended behaviour.

closed 2016-10-12 by eelcocramer