eelcocramer/node-bluetooth-serial-port #109: Fail to install on Windows 8.1

tzvetelin-vassilev, 2016-09-13:
Hey there, I am getting errors when I try and run npm install bluetooth-serial-port. --msvs_version=2012 also don't helps.

I have Python 2.7.12 and installed Visual Studio 2012. The following errors appeared:

> bluetooth-serial-port@1.3.1 install C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port
> node-gyp configure build

C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" configure build )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" configure build ) 
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  BluetoothSerialPort.cc
  DeviceINQ.cc
  BTSerialPortBinding.cc
  BluetoothHelpers.cc
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(336): error C2988: unrecognizable template declaration/definition (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(336): error C2059: syntax error : 'using' (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(469): error C2988: unrecognizable template declaration/definition (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(469): error C2059: syntax error : 'using' (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(576): error C2061: syntax error : identifier 'WeakCallbackType' (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(642) : see reference to class template instantiation 'v8::PersistentBase<T>' being compiled
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(637): error C2253: 'PersistentBase<T>' : pure specifier or abstract override specifier only allowed on virtual function (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(844): error C2253: 'Global<T>' : pure specifier or abstract override specifier only allowed on virtual function (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(847) : see reference to class template instantiation 'v8::Global<T>' being compiled
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(852): error C2988: unrecognizable template declaration/definition (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(852): error C2059: syntax error : 'using' (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(915): error C2989: 'v8::HandleScope' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(319) : see declaration of 'v8::HandleScope'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(949): error C2989: 'v8::EscapableHandleScope' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(135) : see declaration of 'v8::EscapableHandleScope'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(979): error C2989: 'v8::Data' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(74) : see declaration of 'v8::Data'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(1118): error C2989: 'v8::Script' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(96) : see declaration of 'v8::Script'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(1559): error C2989: 'v8::StackTrace' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(101) : see declaration of 'v8::StackTrace'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(1622): error C2989: 'v8::StackFrame' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(100) : see declaration of 'v8::StackFrame'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(2031): error C2989: 'v8::Value' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(108) : see declaration of 'v8::Value'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(2037): error C2989: 'v8::Primitive' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(93) : see declaration of 'v8::Primitive'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(2052): error C2989: 'v8::Boolean' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(70) : see declaration of 'v8::Boolean'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(2072): error C2989: 'v8::Name' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(86) : see declaration of 'v8::Name'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(2439): error C2989: 'v8::String' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(102) : see declaration of 'v8::String'
c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(2477): error C2989: 'v8::Symbol' : class template has already been declared as a non-class template (..\src\windows\BluetoothSerialPort.cc) [C:\Users\Grigor\node_projects\Inkspace\node_modules\bluetooth-serial-port\build\BluetoothSerialPort.vcxproj]
          c:\users\grigor\.node-gyp\4.5.0\include\node\v8.h(104) : see declaration of 'v8::Symbol'
[... cut ...]
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp configure build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs bluetooth-serial-port
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!
npm ERR!     npm owner ls bluetooth-serial-port
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\Grigor\node_projects\Inkspace\npm-debug.log

eelcocramer, 2016-09-14:
Hi, thanks for your report. Were you able to install this module before? Version 1.3.1 was just released yesterday and I wonder if this is a regression.

For the automated builds MVS2015 is used.

You might want to try to compile the module for the x86 architecture instead of x64.

tzvetelin-vassilev, 2016-09-14:
On MacOS everything works fine. I'll use it from there.

closed 2016-09-14 by tzvetelin-vassilev