eelcocramer/node-bluetooth-serial-port #111: BluetoothSerialPortServer Segmentation Fault on Raspberry Pi 3

ollym, 2016-09-19:
Running the following:

```
var server = new(require('bluetooth-serial-port')).BluetoothSerialPortServer();
var PORT = 1;
var UUID = '38e851bc-7144-44b4-9cd8-80549c6f2912';
server.listen(UUID, PORT, (clientAddress) => {
  console.log('Client: ' + clientAddress + ' connected!');
});
```

On a Raspberry Pi 3 causes a Segmentation Fault. I'll try to debug it a little further.

dizag, 2016-09-19:
same to me :(

ollym, 2016-09-19:
So I got this to work, at least not seg fault, by following this:
https://bbs.archlinux.org/viewtopic.php?id=201672

Still working on actually getting it discoverable though.

ollym, 2016-09-20:
So I assume @atilag wrote the lib using the old BlueZ 4 bindings that were removed in the latest BlueZ 5, unless you use the `--compat` flag. The follow works for me getting this back up and running:

```bash
echo "Patching BlueZ 5 service file"
sed -i 's/^ExecStart=\/usr\/lib\/bluetooth\/bluetoothd$/ExecStart=\/usr\/lib\/bluetooth\/bluetoothd --compat/g' /etc/systemd/system/dbus-org.bluez.service
sed -i 's/^ExecStart=\/usr\/lib\/bluetooth\/bluetoothd$/ExecStart=\/usr\/lib\/bluetooth\/bluetoothd --compat/g' /lib/systemd/system/bluetooth.service

echo "Reloading Bluetooth Service..."
systemctl daemon-reload
systemctl restart bluetooth
service bluetooth restart
```

Only thing I'm struggling now with is discovering the device from my laptop

atilag, 2016-09-20:
Yes, I used the API for BlueZ 4 so you need the --compat flag.
I'll try to migrate it to BlueZ 5 D-Bus oriented API for next releases... Seems like a good point updating the doc in the meantime.

dizag, 2016-09-20:
for great regret the **ollym** clue  didn't finish the job (error isn't exist) but... BluetoothSerialPort is still unusable on pi 3 b+

atilag, 2016-09-20:
@dizag I used a RPI 3 for the development and it worked like a charm. What's exactly your problem?
Is it segfaulting? Cannot discover the device?...
Make sure that bluetoothd is running with the --compat flag:
`ps aux | grep bluetoothd`

dizag, 2016-09-20:
hey, 
i need to connect my app to pi3b nodejs ... through bluetooth (but not BLE) ... and every article say BluetoothSerialPort is my tool :) 
first i follow http://pratyeka.org/rfcomm/ and I am absolutely sure about bluetooth is on and comunicate
...
ps aux | grep bluetoothd
root      2795  0.0  0.4   5128  3580 ?        Ss   17:05   0:00 /usr/lib/bluetooth/bluetoothd --compat
root      3100  0.0  0.2   4248  1664 pts/3    S+   17:35   0:00 grep bluetoothd

so after two days digging what is problem with BluetoothSerialPort i reinstall raspbian(from jessy) -> noobs
and now on
    server.advertise(UUID, function(err) {
        if (err) {
            console.log('Cannot advertise the server');
        }
    });
give me **server.advertise isn't a function**
...

before **ollym** clue was error Segmentation Fault on
    server.listen(UUID, PORT, function (clientAddress) {
        console.log('Client: ' + clientAddress + ' connected!');
        server.on('data' function(buffer) {
            console.log('Received data from client: ' + buffer);
        });
    });
now there haven't error but still didn't work

atilag, 2016-09-20:
server.advertise() is not a method anymore. Advertising is automatically achieved by calling listen()... so yes, I need to update the README.md ASAP and remove the code example!

dizag, 2016-09-20:
yep, thanks ...

can you give me short example, (because my app is builded by as3/air/ANE and i don't know exact chanel) ... 
first : how to convert my mac(B8:27:EB:87:E5:21) to UUID 
and 
second how to discover channel/port

thanks in advance :)

ollym, 2016-09-20:
@atilag another question I have is how the UUID is important? If this lib exposes both a client and a server, how would I get the UUID of the service through the client?

ollym, 2016-09-20:
@atilag I also have trouble with a device connecting to this service.

If I run `sdptool browse local` after starting a server using the exact same code at the top of this issue, I can see your service entry here:

```
Service Name: RFCOMM Server socket
Service Description: A RFCOMM listening socket
Service Provider: Service Provider
Service RecHandle: 0x10005
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: 1
```

However this doesn't work for me, I don't get any connection event when my device tries to connect, (actual BT pairing works fine). However, in exactly the same environment, if instead I do:

```
$ sdptool add SP
$ rfcomm watch /dev/rfcomm0 1 /sbin/agetty rfcomm0 921600 linux
Waiting for connection on channel 1
```

After a little while after pairing, I get a connection:

```
Connection from 00:12:6A:F6:0D:57 to /dev/rfcomm0
Press CTRL-C for hangup
```

Again, after looking at the `sdptool browse local` output:

```
Service Name: Serial Port
Service Description: COM Port
Service Provider: BlueZ
Service RecHandle: 0x10006
Service Class ID List:
  "Serial Port" (0x1101)
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: 1
Language Base Attr List:
  code_ISO639: 0x656e
  encoding:    0x6a
  base_offset: 0x100
Profile Descriptor List:
  "Serial Port" (0x1101)
    Version: 0x0100
```

It's clear there are some differences between your service and the BlueZ one. My device works with BlueZ but not yours. (For reference, the device I'm using is http://www.opticon.com/PX-20.aspx) which you can configure to connect to a given BT device as a slave.

Specifically I think you're missing the `Serial Port` in the `Service Class ID List`. But it would probably be good practice to add the `Language Base Attr` fields too.

I'll see whether I can create a pull request to add this myself and test locally.

ollym, 2016-09-20:
Haha so I managed to answer my first question with my second. So to get the UUID working with another serial port you have to use the Serial Port UUID, which happens to be `0x1101` like so:

```node
var server = new(require('bluetooth-serial-port')).BluetoothSerialPortServer();
var PORT = 1;
var UUID = '1101';
server.listen(UUID, PORT, (clientAddress) => {
  console.log('Client: ' + clientAddress + ' connected!');
});
```

This now works perfectly with my device.

@atilag this is a serial port specific library, so I think it's important to make 0x1101 the default parameter, and perhaps allow people to override it in an optional settings menu. That would mean then this same library could be used on both the client and server devices without additional configuration.

Case closed.

closed 2016-09-20 by ollym