eelcocramer/node-bluetooth-serial-port #113: Adds defaults for UUID and channel in server

eelcocramer, 2016-09-21:
I think it will improve the server code if we add sane defaults for UUID and channel so errors / mistakes like mentioned in #111 will not occur anymore.

I see 3 approaches to this. Both will break the backward compatibility on the API tough.

1. Set a default for UUID and channel in the constructor of `BluetoothSerialPortServer` and pass in an optional `opts` object to override the defaults. The `uuid` and `listeningChannel` than can be removed from the `listen` function.

    > This has a side effect though. Now a `BluetoothSerialPortServer` object can be used to listen to multiple channels / uuid tuples. With the above proposal this will become impossible.

2. Change the signature of the `listen` function to `listen(successCb, errorCb, uuid, listeningChannel)`. The implementation could be changed to something like:

    ```javascript
    BluetoothSerialPortServer.prototype.listen = function () {
       var uuid = 1101, channel = 1, successCallback = arguments[0], errorCallback = undefined;

       if (typeof arguments[1] === 'function') {
           errorCallback = arguments[1];
           if (arguments.length > 2) {
               uuid = arguments[2];
               channel = arguments[3];
           }
       } else {
           // you get the point...
       }

       // ....
    }
    ```

    This probably needs some sanity checks but you'll get the point.

3. Change the signature of the `listen` function to `listen(opts)`. Were the `opts` object must contain a fields for `successCb` and optionally could contain fields for `errorCb`, `uuid` and `channel`.

4. Change the signature of the `listen` function to `listen(successCb, errorCb, opts)` were both `errorCb` and `opts` would be optional fields so these calls should all be valid: `listen(cb)`, `listen(cb, opts)`, `listen(cb, ecb)` or `listen(cb, ecb, opts)`.

What do you think of any of these approaches @atilag? Personally I would prefer 1 or 4. Depending if listening to multiple channels / uuids should be supported or not.

atilag, 2016-09-21:
As @ollym said in #111 , makes sense to default the UUID to 0x1101 ([SPP](https://www.bluetooth.com/specifications/assigned-numbers/service-discovery)) because after all this is a Bluetooth Serial Port library. The ideal scenario would be make the port/channel selection optional, so the library will choose the first one available but this is not critical so we can propose this as an improvement for later releases. 
I will go for changing the signature of the listen() method (option 4), that will break compatibility but I guess is something our users would understand :) ... better fix design problems now that we are in "beta" for Server Side code.

eelcocramer, 2016-09-27:
done in `2.0.0`

closed 2016-09-27 by eelcocramer