eelcocramer/node-bluetooth-serial-port #115: BluetoothSerialPortServer does not accept re-connection after port closed

ollym, 2016-09-22:
As it says on the tin!

ollym, 2016-09-22:
Also have you done any testing listening on multiple channels? I'm having trouble with that too.

ollym, 2016-09-23:
So my example code to try and fix the problem:

```node
  const listen = (channel) => {
    let server = new BluetoothSerialPortServer;
    console.log(`listening for devices on channel ${channel}!`);

    server.listen('1101', channel, (address) => {
      console.log(`found new device at ${address} on channel ${channel}!`);

      server.on('closed', () => {
        console.log(`device ${address} disconnected on channel ${channel}!`);
        listen(channel);
      });
    });
  };

  listen(1);
```

The output:

```
listening for devices on channel 1!
found new device at 00:12:6A:F6:0D:57 on channel 1!
scanner 00:12:6A:F6:0D:57 disconnected on channel 1!
listening for scanners on channel 1!
Illegal instruction (core dumped)
```

Guess you can't listen on a port you were already listening on.

eelcocramer, 2016-09-23:
I would assume that the server keeps listening and the `closed` event is only emitted to let the server code know that an remote BT devices disconnected. Did you try reconnecting after a `close` (without calling `listen` again on the server?

ollym, 2016-09-23:
Yes I did, but had no luck.

ollym, 2016-09-23:
So the problem appears to be that after a connection, the socket is removed from SDP:
https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/linux/BTSerialPortBindingServer.cc#L583-L587

And isn't re-opened afterwards when the socket closes.

ollym, 2016-09-23:
My only work around so far is to do this:

```node
  const listen = (channel) => {
    let server = new BluetoothSerialPortServer;
    console.log(`listening for scanners on channel ${channel}!`);
    server.listen('1101', channel, (address) => {
      listen(channel + 1);

      console.log(`found new scanner at ${address} on channel ${channel}!`);
      emitter.emit('connected', new PX20(server));

      server.on('closed', () => {
        console.log(`scanner ${address} disconnected on channel ${channel}!`);
      });
    });
  };

  listen(1);
```

It's a terrible solution, @atilag please work your magic when you get a chance.

atilag, 2016-09-25:
The API is no ready to listen in multiple channels at the same time (by calling listen() sequentially, for example). Sounds like something I could try for next releases, thought :)
Haven't look into the code yet to see why is segfaulting, but it should be failing in a controlled way because you can't listen in a channel that's still open (or is being closed by the kernel), that is something I could control in the library side of course. For next releases, avoiding to set a channel number will end up looking through all the channels until a free one is found (unfortunately, I read that there's no such functionality in BlueZ).
I'm gonna take a look at the closing procedures and see where's the problem...
Thanks for reporting @ollym !!! :)

ollym, 2016-09-25:
@atilag would it not be better to just re-advertise the channel after it's disconnected and so any new connection just gets called into the same callback? I don't think it's intuitive to have to re-listen unless you've explicitly it.

andreavaresio, 2017-01-03:
Hi I would like to know whether a solution has been found to this problem.

I Am trying the following and the second time the client disconnects, an error occurred.

`var server = new(require('bluetooth-serial-port')).BluetoothSerialPortServer();
var CHANNEL = 1; // My service channel. Defaults to 1 if omitted.
//var UUID = '38e851bc-7144-44b4-9cd8-80549c6f2912'; // My own service UUID. Defaults to '1101' if omitted
var UUID = '1101';

function HandleAConnection()
{
	console.log('Waiting for connection...');
	server.listen(function (clientAddress) {
		console.log('Client: ' + clientAddress + ' connected!');
		server.on('data', function(buffer) {
			console.log('Received data from client: ' + buffer);

			// ...

			console.log('Sending data to the client');
			server.write(new Buffer('ciao'), function (err, bytesWritten) {
				if (err) {
					console.log('Error!');
				} else {
					console.log('Send ' + bytesWritten + ' to the client!');
				}
			});
		});
	}, function(error){
		console.log("Something wrong happened!:" + error);
	}, {uuid: UUID, channel: CHANNEL} );
	server.on('closed', function(){
		console.log('Client: disconnected!');
		HandleAConnection();
	});
}
HandleAConnection();
`

output is 

Waiting for connection...
Client: 44:D4:E0:7B:E5:94 connected!
Client: disconnected!
Waiting for connection...
Client: 44:D4:E0:7B:E5:94 connected!
Client: disconnected!
Waiting for connection...
Client: disconnected!
Waiting for connection...
Something wrong happened!:Error: Couldn't bind bluetooth socket. errno:98

andreavaresio, 2017-01-03:
It seems that 'closed' events arrives twice and thus server goes into error.

andreavaresio, 2017-01-03:
further investigation: every time a client disconnects, an additional 'closed'event is fired.

I had this warning that seems to confirm the bug

 Warning: Possible EventEmitter memory leak detected. 11 data listeners added. Use emitter.setMaxListeners() to increase limit

atilag, 2017-01-03:
Ok, let me take a look :)

closed 2018-04-12 by eelcocramer