eelcocramer/node-bluetooth-serial-port #116: bluetooth-serial-port && Electron

amrish7, 2016-09-27:
`bluetooth-serial-port` module is not connecting to a paired device when I try using it within Electron framework. I am not seeing a success/error response to my `btSerial.connect` API call

Wondering if this is a known issue and/or is there a plan to add support for popular frameworks like Electron? Here is the code snippet that works fine from node, but *not* on Electron

```
var Bluetooth = require('bluetooth-serial-port');
var btSerial = new Bluetooth.BluetoothSerialPort();

const deviceEnum = (devs) => {
    console.log(`${devs.length} BlueTooth device(s) are connected.`);
    devs.forEach((dev) => {
        console.log(`Paired with "${dev.name}"`);

        if(dev.name.indexOf('<MY-DEVICE-NAME> ') === 0) {
            console.log(`Trying to connect with ${dev.name}`);
            btSerial.connect(dev.address, 1, () => {
                console.log(`CONNECTED >>> `);
            }, (e) => {
                console.error(`ERROR ON CONNECTED ${e}`);
            });
        }
    });
};
btSerial.listPairedDevices(deviceEnum);
```

**Environment information**
   * node `v 6.5.0`
   * electron `v 1.4.1` (Chrome `v 53.0.2785.113`
   * OS X `v 10.11.6`

eelcocramer, 2016-09-28:
Please check FAQ issue #73

amrish7, 2016-09-28:
@eelcocramer If I am not wrong, the #73 talks about module version mismatch and it is not the issue I am not referring to. I did get over the version mismatch issue by a simple uninstall and reinstall, but this issue is a bit deeper I think... I am able to get a list of paired devices, but unable to connect to them when running the App in Electron environment

eelcocramer, 2016-09-28:
There was also a discussion about electron in #100. I hope that helps. I've no current plans to add support for Electron myself, but I would be happy to accept a PR if needed.

BlaiseGratton, 2017-03-17:
@amrish7 I ran into the same issue and resolved it by rebuilding the node module to target your version of Electron and CPU architecture. For example, for Electron 1.2.5, `cd` into `node_modules/bluetooth-serial-port` and run `node-gyp rebuild --target=1.2.5 --arch=x64 --dist-url=https://atom.io/download/atom-shell`. Hope this helps

closed 2017-03-20 by eelcocramer