eelcocramer/node-bluetooth-serial-port #117: BluetoothSerialPortServer not being loaded when require("bluetooth-serial-port") is called

pjrader1, 2016-09-27:
Unless I am doing something entirely wrong, BluetoothSerialPortServer isn't being loaded with require("bluetooth-serial-port"), only BluetoothSerialPort is. I copy-pasted the example code from the README, and tried to change it to work:

```
var BluetoothSerialPortServer = require("bluetooth-serial-port").BluetoothSerialPortServer;
var server = new BluetoothSerialPortServer();                                 

var UUID = "";

server.listen(function(clientAddress) {
    console.log("Client: " + clientAddress + " connected!");
    server.on("data", function(buffer) {
        // TODO
    });
}, function(error) {
    console.error("Something wrong happened!: " + error);
}, {uuid: UUID});
```

When I try to run the code, this error occurs:
```
TypeError: BluetoothSerialPortServer is not a function
```

I printed out the object for require("bluetooth-serial-port"), and this is what came out:
```
{ BluetoothSerialPort: 
   { [Function: BluetoothSerialPort]
     super_: 
      { [Function: EventEmitter]
        EventEmitter: [Circular],
        usingDomains: false,
        defaultMaxListeners: 10,
        init: [Function],
        listenerCount: [Function] } } }
```

From the looks of it, BluetoothSerialPortServer is not being loaded when require("bluetooth-serial-port") is called. I used npm to install the library, and looked at the javascript file being included. It is what it should be, at least that's in the repo. I don't know how node modules are programmed, so I really don't know why this is happening.

eelcocramer, 2016-09-28:
Are you running this on Linux?

pjrader1, 2016-09-28:
I’m running this on macOS.

eelcocramer, 2016-09-28:
The server is currently only supported on Linux.

eelcocramer, 2016-09-28:
Where I wrote supported I actually mean implemented. Sorry.

pjrader1, 2016-09-28:
@eelcocramer Okay. I'll close this issue. The README should make that clear. Right now it looks like the server supports all platforms.

closed 2016-09-28 by pjrader1

eelcocramer, 2016-09-28:
There is a note in the readme about this but it could me made more clear I assume.