eelcocramer/node-bluetooth-serial-port #120: Error installing on node v6.8.1

simonl65, 2016-10-16:
Was trying to install Cardash using npm install, got:

npm ERR! Linux 4.1.14-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v6.8.1
npm ERR! npm  v3.10.8
npm ERR! code ELIFECYCLE

npm ERR! bluetooth-serial-port@1.3.1 install: `node-gyp configure build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bluetooth-serial-port@1.3.1 install script 'node-gyp configure build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the bluetooth-serial-port package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp configure build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs bluetooth-serial-port
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls bluetooth-serial-port
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/Cardash/npm-debug.log

eelcocramer, 2016-10-16:
Can you post the `npm-debug.log` (mentioned in the last line of the log you posted)?

simonl65, 2016-10-16:
My mistake - I hadn't installed build-essential. I'll test again tomorrow.

closed 2016-10-16 by simonl65

eelcocramer, 2016-10-16:
Thanks for letting me know. Good luck!