eelcocramer/node-bluetooth-serial-port #122: Questions about UUID & listPairedDevices

tzvetelin-vassilev, 2016-10-28:
1. How to connect to device with particular UUID? I would like to filter devices somehow and UUID looks like a way to do this.
2. listPairedDevices looks working on windows partially, what is the problem there? If this is fixable it would be great if inquiry process reports and services info.

eelcocramer, 2016-10-28:
To answer 1: you can check if a found device matches a particular UUID (or name) when the inquire emits the `found` event. In [this example](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/experiments/scan-test.js#L26) you can see how the code filters on a particular device name after searching for the serial port channel (as this example lists all available serial port and devices before connecting to a particular one). In the example the `UUID` can be found in the `address` variable. Please note that an `UUID` can be formatted differently depending on the platform you are using. Some platforms separate the bytes of the `UUID` with colons others use dashes. 

As for you other question: @ELundby45 worked on that implementation. Maybe he is able to answer that question?

tzvetelin-vassilev, 2016-10-28:
Well, address variable is 6 bytes and this is the address of the device. UUID is 16 bytes and is something different. I suppose this is an id of some service and could be unique identifier for the device.

eelcocramer, 2016-10-28:
Sorry, you are right about the UUID is different to the address of the device.

The UUID identifies the service. This module looks for the Serial Port Profile service so the UUID is hardcoded to that. If you want to identify a device you should use the address.

ELundby45, 2016-10-28:
Let me know if I can be of assistance. But I think you covered it eelcocramer. 

Since this library targets the Serial Port Profile (SPP) service, listPairedDevices doesn't do service inquiry and simply returns the channel for the SPP service and the address of the device. 

If this doesn't answer your question, @tzvetelin-vassilev could you clarify further?

tzvetelin-vassilev, 2016-10-31:
Let me try to explain what I'm trying to do. I'm trying to find device from a device line. Name is update-able and could not be something meaningful. Address is 6 bytes and if is possible to extract VendorID and ProductID I wouldn't disturb you. I'm not sure this is possible so I need something more to recognize the device. So my questions are:

1. Do you have access to services provided by device while inquiry before pairing. If it is possible to take them, found event could return them as third parameter.
2. Under MacOS I can see a lot of services provided by device. What is the problem with Windows and is this list is possible to be provided and there?
3. Service ID (UUID) could be exposed in this list?
4. Why only SPP? It would be great to connect to any service with UUID as parameter.

eelcocramer, 2016-10-31:
Currently the library only searches for SPP because that is what this module does, provide serial port access over Bluetooth. You can take the discovery mechanism and add you own services if you want.

I'm happy to accept a PR that makes the inquire function more generic and makes the UUID available to the calling application.

From the top of my head I do not think a device needs to be paired to use SDP so I believe it should be possible to create a function that returns the available services before pairing.

closed 2018-04-12 by eelcocramer