eelcocramer/node-bluetooth-serial-port #123: cannot receive data on osx, strange as sending data works

gengshenghong, 2016-11-08:
is there any trick to use this library on os x ? In my simple test code, it can send over Bluetooth and target received, but cannot receive back! Same code work on linux and windows. code is just helloworld code to send and listen on data event, target is arduino to send back what received.

any special settings needed on mac? I have tested on osx 10.10 and 10.11, latest node 6.9.1 version, mac mini and pro....

gengshenghong, 2016-11-08:
@eelcocramer any idea? can you share me your dev/test environment for osx? Thanks very much.

eelcocramer, 2016-11-08:
Nothing special needed. There is an example in `experiments/scan-test.js` that I've used on OSX for testing.

gengshenghong, 2016-11-09:
@eelcocramer Thanks for the information. To debug the issue, I've installed Hackintosh  in vmware, and found below behaviour:
1. using USB2.0 for the bluetooth adapter, it works.
2. using USB3.0 for the bluetooth adapter, it can send successfully (and target received), but cannot receive.
3. You mentioned Xcode is required, but in my environment, I install bluetooth-serial-port without Xcode, I do see an error message but it installed successfully and USB 2.0 works well! Maybe this is the reason of the strange issue? Maybe it should block the installation if this is the root cause. I'll install XCode to confirm this and update here. It is really strange as 80% of the functionalities works (can scan, connect, send, but cannot receive only!)

npm install bluetooth-serial-port results as below:
TestdeMac:node_test test$ npm install bluetooth-serial-port  --registry=https://registry.npm.taobao.org

> bluetooth-serial-port@2.1.0 install /Users/test/Desktop/node_test/node_modules/bluetooth-serial-port
> node-gyp configure build

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/BluetoothWorker.o
../src/osx/BluetoothWorker.mm:359:57: warning: 'getAddressString' is deprecated:
      first deprecated in OS X 10.7 [-Wdeprecated-declarations]
        [self disconnectFromDevice: [[rfcommChannel getDevice] getAddres...
                                                               ^
/System/Library/Frameworks/IOBluetooth.framework/Headers/objc/IOBluetoothDevice.h:477:1: note: 
      'getAddressString' has been explicitly marked deprecated here
- (NSString *)getAddressString DEPRECATED_IN_MAC_OS_X_VERSION_10_7_AND_LATER;
^
../src/osx/BluetoothWorker.mm:380:35: warning: 'getAddressString' is deprecated:
      first deprecated in OS X 10.7 [-Wdeprecated-declarations]
                        strcpy(info->address, [[device getAddressString...
                                                       ^
/System/Library/Frameworks/IOBluetooth.framework/Headers/objc/IOBluetoothDevice.h:477:1: note: 
      'getAddressString' has been explicitly marked deprecated here
- (NSString *)getAddressString DEPRECATED_IN_MAC_OS_X_VERSION_10_7_AND_LATER;
^
../src/osx/BluetoothWorker.mm:381:32: warning: 'getNameOrAddress' is deprecated:
      first deprecated in OS X 10.6 [-Wdeprecated-declarations]
                        strcpy(info->name, [[device getNameOrAddress] UT...
                                                    ^
/System/Library/Frameworks/IOBluetooth.framework/Headers/objc/IOBluetoothDevice.h:448:1: note: 
      'getNameOrAddress' has been explicitly marked deprecated here
- (NSString *)getNameOrAddress DEPRECATED_IN_MAC_OS_X_VERSION_10_6_AND_LATER;
^
3 warnings generated.
  CC(target) Release/obj.target/BluetoothSerialPort/src/osx/pipe.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/BluetoothDeviceResources.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/BluetoothSerialPort.o
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/BTSerialPortBinding.o
../src/osx/BTSerialPortBinding.mm:190:57: warning: 'NewInstance' is deprecated
      [-Wdeprecated-declarations]
        Local<Object> resultBuffer = bufferConstructor->NewInstance(1, c...
                                                        ^
/Users/test/.node-gyp/6.9.1/include/node/v8.h:3270:31: note: 'NewInstance' has
      been explicitly marked deprecated here
                Local<Object> NewInstance(int argc, Local<Value> argv[]) const);
                              ^
/Users/test/.node-gyp/6.9.1/include/node/v8config.h:336:3: note: expanded from
      macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
  ^
1 warning generated.
  SOLINK_MODULE(target) Release/BluetoothSerialPort.node
  SOLINK_MODULE(target) Release/BluetoothSerialPortServer.node
/Users/test/Desktop/node_test
└─┬ bluetooth-serial-port@2.1.0 
  ├── bindings@1.1.1 
  └── nan@2.4.0 

npm WARN enoent ENOENT: no such file or directory, open '/Users/test/Desktop/node_test/package.json'
npm WARN node_test No description
npm WARN node_test No repository field.
npm WARN node_test No README data
npm WARN node_test No license field.
TestdeMac:node_test

Thanks,
Shenghong

eelcocramer, 2016-11-09:
From the logs it look like the module is build fine using the XCode commandline tools. From what I read it can be a driver issue on your USB3.0 device. Maybe you should check this with the people that created the driver for this device?

gengshenghong, 2016-11-12:
Hi @eelcocramer 

I've spent several days debugging the issue, and please forgot about the USB 2.0 and 3.0 stuff, it was a mistake I made to use VMWare. Whatever, I've NEVER succeeded on a Mac hardware machine (and in VMWare now, the previous success is by mistake).

Could you please share more details about your dev environment, like OS X version and Node version?

I've got below results now (all are setup well, xcode is installed too), after many times confirmation. And it do seems like an issue on Max OS X side.
1. I am using Bluetooth to talk with an Arduino board. The same code works on Windows and Linux (Thanks for this great library). But only fail on Mac OS now.
2. With some debugging, the test case is simplified as below:
Arduino Side (Arduino is simple, it will first run setup and then run like: while(1) {loop()}
`
void setup() {
  Serial.begin(57600);
}

void loop() {
  if(Serial.available()) { // this will be triggered when connected
    delay(100); // delay 100ms  // if no delay after connected here, or delay less than 15ms, it works well on Max OS X side???
    Serial.write(0x55);
    Serial.read();
  }
}
`
See the comments above, if my Arduino write something back to Mac OSX over bluetooth after connection, your library works well. If there is a very short delay, the library does not work for me.

The code on OSX side is also simple, it just connect, write and listen on data. When it does not work (delay in Arduino), "data" event will never come.
`
/*
 // npm install bluetooth-serial-port
 */
var btSerial = new (require('bluetooth-serial-port')).BluetoothSerialPort();

var address = '00-ba-55-57-5d-fd';

var channel = 1;
console.log("channel:" + channel);
btSerial.connect(address, channel, function() {
  console.log('connected');
  console.log('write');
  btSerial.on('data', function(buffer) {
    console.log("data");
    console.log(buffer);
  });
  btSerial.write(new Buffer([0x01, 0x02, 0x03]), function(err, bytesWritten) {
    if (err) console.log(err);
  });

  btSerial.write(new Buffer([0x04, 0x05, 0x06), function(err, bytesWritten) {
    if (err) console.log(err);
  });
  }, function () {
  console.log('cannot connect');
});

console.log("start inquire...");
btSerial.inquireSync();
console.log("inquire done...");
`

Any idea? I appreciate if you can give me some hints or help a test on your side. I've tried to debug your OSX object C code, but as I have zero experience on Mac, it is difficult for me.

Thanks,
Shenghong

gengshenghong, 2016-11-12:
I've reproduced the same issue using obj-C (though it is my first time to write obj-c code...).

Below are the objective C code:
`
#import <Foundation/Foundation.h>
#import <IOBluetooth/objc/IOBluetoothDevice.h>          // Note the import for bluetooth
#import <IOBluetooth/objc/IOBluetoothDeviceInquiry.h>   // Note the import for bluetooth
#import <IOBluetooth/objc/IOBluetoothRFCOMMChannel.h>

@interface Test:NSObject{
 IOBluetoothDevice* btDevice;
}
-(bool) openConnection; //:(BluetoothRFCOMMChannelID *) chanId;

- (void)rfcommChannelData:(IOBluetoothRFCOMMChannel*)rfcommChannel data:(void *)dataPointer length:(size_t)dataLength;
@end

@implementation Test

-(bool) openConnection //: (BluetoothRFCOMMChannelID *) chanId
{
    IOReturn status;
    
    NSString *address = @"00-ba-55-57-5d-fd";
    NSNumber *channelID = @1;
    
    IOBluetoothDevice *device = [IOBluetoothDevice deviceWithAddressString:address];
    
    // open the baseband connection...
    status = [device openConnection];
    if(status == kIOReturnSuccess) {
        NSLog(@"connect baseband success...");
    } else {
        NSLog(@"connect baseband fail...");
    }
    
    IOBluetoothRFCOMMChannel *channel = [[IOBluetoothRFCOMMChannel alloc] init];
    
    if ([device openRFCOMMChannelSync: &channel withChannelID:[channelID intValue] delegate: self] == kIOReturnSuccess) {
        NSLog(@"connect success.");
    } else {
        NSLog(@"connect fail.");
    }
    return YES;
}

- (void)rfcommChannelData:(IOBluetoothRFCOMMChannel*)rfcommChannel data:(void *)dataPointer length:(size_t)dataLength
{
    NSData *data = [NSData dataWithBytes: dataPointer length: dataLength];
    
    NSLog(@"data...");
    NSLog(@"%zu", dataLength);
    NSLog(@"%@", data);
}

@end

int main(int argc, const char * argv[])
{
    @autoreleasepool {
        Test *a = [[Test alloc] init];
        bool x = [ a openConnection ];
        //[ a openConnection ];
        
        [[NSRunLoop currentRunLoop] run];
        
         //NSLog(@"wait");
        //[NSThread sleepForTimeInterval: 5];
       // NSLog(@"done");
    }
    return 0;
}
`

Arduino will send back data when connected automatically. If I have a delay in Arduino side, OS X cannot receive any data (rfcommChannelData not called)! Or, it works well. This blocks me as it is hard to make sure Arduino will send back data immediately when connection comes and it does not make sense too. I have no idea but it seems like an issue of OS X API now! :(

Thanks,
Shenghong

eelcocramer, 2016-11-12:
Are you sure that your Arduino code does what you expect? In the sample above you are only sending data when data is received from the serial port. `[Serial.available()](https://www.arduino.cc/en/Serial/Available)` only becomes `true` when data is available to read. So when you are not sending to the Arduino is will not send any data out either.

gengshenghong, 2016-11-12:
it will send data. the Bluetooth module will send a string "connect" when connected.

eelcocramer, 2016-11-12:
You might want to check this [gist](https://gist.github.com/eelcocramer/4117108) that has the original test code I used for all platforms from [this post](https://servicelab.org/2012/12/12/wirelessly-control-your-arduino-with-nodejs-over-bluetooth/)

gengshenghong, 2016-11-12:
thanks for confirm. great that you have tested on arduino too. if it works for you, it might be an compatible issue of my slave Bluetooth module, which is a simplified hc05/hc06 compatible slave-only module?
i will find a hc05 module to have a try. :(

gengshenghong, 2016-11-19:
@eelcocramer Some updates here. I've got an official-HC06 module to test this issue.
1. with HC06, the connect will always fail for the first time, and only connect successfully if I connect tries in the code. Also, I've tested with obj-C and same behavior. Are you aware of this?
2. with HC06 (I connect twice), the above issue disappear now. It works well to send and receive now. So, this seems like an issue of my Bluetooth slave module.

Let me know if you have any ideas why it must conect twice (Mac OS 10.11).

Thanks,
Shenghong

eelcocramer, 2016-11-25:
Resolved as mentioned in #125

closed 2016-11-25 by eelcocramer