eelcocramer/node-bluetooth-serial-port #125: hang or crash at write (write call back not called) on osx sometimes

gengshenghong, 2016-11-25:
Hi @eelcocramer 

After #123 resolved, I've run into other issues. On Mac OSX, it is not stable and the write function will hand sometimes, I cannot work around this. After some debugging, I've got some progress, but as I am not familiar with obj-c at all, I'll post here and hope you can help.

First of all, the logic on nodejs code side is simple, I will do like below:
1. connect
2. write
3. on the write callback, write another data (or write after some time)
4. it will hang sometimes
(it does not happen always, when it works, I've run the write logic for even 1 hour and never crash. when it does not work, it stop at less than 100 times write, and usually the crash as almost same times of writing!)

To debug where it hangs, I've added some log in your library and find that it hangs at getInstance of BluetoothWorker, more specifically, it hangs at objectForKey (which seems like a simple function?).

See below information:
In BluetoothWorker.mm, below logs are added:
/** The BluetoothWorker class is a singleton. An instance can be obtained using this method */
+ (id)getInstance: (NSString *) address
{
    NSLog(@"getInstance..");
    if (!address) {
        address = @"inquiry";
    }
    //NSLog(address);

    static NSMutableDictionary *instances = nil;
    static dispatch_once_t onceToken;

    NSLog(@"get 1");
    dispatch_once(&onceToken, ^{
        instances = [[NSMutableDictionary alloc] init];
        globalConnectLock = [[NSLock alloc] init];
        instanceWorkers = [[NSMutableDictionary alloc] init];
    });
    if (instances == nil) {
        NSLog(@"nil...");
    }
    NSLog(@"get 2");

    BluetoothWorker *worker = nil;
    NSLog(address);
    @synchronized (instances) {
        NSLog(@"get 3");
        worker = [instances objectForKey:address]; // it hand at this line, why????
        NSLog(@"get 4");
        if (worker == nil) {
            worker = [[BluetoothWorker alloc] init];
            [instances setObject: worker forKey: address];
        }
    }
    //NSLog(@"get 5"); // added this line make it very hard to reproduce!!
    return worker;
}

See comments in above code, if I added the line "get 5" log, it is very hard to reproduce. Below are the logs when it hangs:
2016-11-24 21:02:52.806 node[3439:346407] getInstance..
2016-11-24 21:02:52.807 node[3439:346407] get 1
2016-11-24 21:02:52.807 node[3439:346407] get 2
2016-11-24 21:02:52.807 node[3439:346407] 20:16:10:19:11:45
2016-11-24 21:02:52.807 node[3439:346407] get 3
2016-11-24 21:02:52.807 node[3439:346407] get 4
[2016-11-24 21:02:56.258] [INFO] test - Start Time:11/24/2016, 9:02:56 PM
write...
2016-11-24 21:02:56.271 node[3439:346409] EIO_Write
2016-11-24 21:02:56.271 node[3439:346409] EIO_Write1
2016-11-24 21:02:56.271 node[3439:346409] EIO_Write2
2016-11-24 21:02:56.271 node[3439:346409] EIO_Write3
2016-11-24 21:02:56.271 node[3439:346409] EIO_Write4
2016-11-24 21:02:56.271 node[3439:346409] getInstance..
2016-11-24 21:02:56.271 node[3439:346409] get 1
2016-11-24 21:02:56.271 node[3439:346409] get 2
2016-11-24 21:02:56.272 node[3439:346409] 20:16:10:19:11:45
2016-11-24 21:02:56.272 node[3439:346409] get 3
2016-11-24 21:02:56.272 node[3439:346409] get 4
2016-11-24 21:02:56.272 node[3439:346409] EIO_Write5
2016-11-24 21:02:56.272 node[3439:346409] to write...
write callback...
write...
2016-11-24 21:02:56.326 node[3439:346409] EIO_Write
2016-11-24 21:02:56.326 node[3439:346409] EIO_Write1
2016-11-24 21:02:56.327 node[3439:346409] EIO_Write2
2016-11-24 21:02:56.327 node[3439:346409] EIO_Write3
2016-11-24 21:02:56.327 node[3439:346409] EIO_Write4
2016-11-24 21:02:56.327 node[3439:346409] getInstance..
2016-11-24 21:02:56.327 node[3439:346409] get 1
2016-11-24 21:02:56.327 node[3439:346409] get 2
2016-11-24 21:02:56.327 node[3439:346409] 20:16:10:19:11:45
2016-11-24 21:02:56.327 node[3439:346409] get 3
2016-11-24 21:02:56.327 node[3439:346409] get 4
2016-11-24 21:02:56.327 node[3439:346409] EIO_Write5
2016-11-24 21:02:56.327 node[3439:346409] to write...
write callback...
write...
2016-11-24 21:02:56.378 node[3439:346409] EIO_Write
2016-11-24 21:02:56.378 node[3439:346409] EIO_Write1
2016-11-24 21:02:56.378 node[3439:346409] EIO_Write2
2016-11-24 21:02:56.378 node[3439:346409] EIO_Write3
2016-11-24 21:02:56.378 node[3439:346409] EIO_Write4
2016-11-24 21:02:56.378 node[3439:346409] getInstance..
2016-11-24 21:02:56.379 node[3439:346409] get 1
2016-11-24 21:02:56.379 node[3439:346409] get 2
2016-11-24 21:02:56.379 node[3439:346409] 20:16:10:19:11:45
2016-11-24 21:02:56.379 node[3439:346409] get 3
2016-11-24 21:02:56.379 node[3439:346409] get 4
2016-11-24 21:02:56.379 node[3439:346409] EIO_Write5
2016-11-24 21:02:56.379 node[3439:346409] to write...
write callback...
write...
2016-11-24 21:02:56.430 node[3439:346409] EIO_Write
2016-11-24 21:02:56.430 node[3439:346409] EIO_Write1
2016-11-24 21:02:56.430 node[3439:346409] EIO_Write2
2016-11-24 21:02:56.430 node[3439:346409] EIO_Write3
2016-11-24 21:02:56.431 node[3439:346409] EIO_Write4
2016-11-24 21:02:56.431 node[3439:346409] getInstance..
2016-11-24 21:02:56.431 node[3439:346409] get 1
2016-11-24 21:02:56.431 node[3439:346409] get 2
2016-11-24 21:02:56.431 node[3439:346409] 20:16:10:19:11:45
2016-11-24 21:02:56.431 node[3439:346409] get 3
2016-11-24 21:02:56.432 node[3439:346409] get 4
2016-11-24 21:02:56.432 node[3439:346409] EIO_Write5
2016-11-24 21:02:56.432 node[3439:346409] to write...
write callback...
write...
2016-11-24 21:02:56.493 node[3439:346409] EIO_Write
2016-11-24 21:02:56.494 node[3439:346409] EIO_Write1
2016-11-24 21:02:56.494 node[3439:346409] EIO_Write2
2016-11-24 21:02:56.494 node[3439:346409] EIO_Write3
2016-11-24 21:02:56.494 node[3439:346409] EIO_Write4
2016-11-24 21:02:56.494 node[3439:346409] getInstance..
2016-11-24 21:02:56.494 node[3439:346409] get 1
2016-11-24 21:02:56.494 node[3439:346409] get 2
2016-11-24 21:02:56.495 node[3439:346409] 20:16:10:19:11:45
2016-11-24 21:02:56.495 node[3439:346409] get 3
2016-11-24 21:02:56.495 node[3439:346409] get 4
2016-11-24 21:02:56.495 node[3439:346409] EIO_Write5
2016-11-24 21:02:56.495 node[3439:346409] to write...
write callback...
write...
2016-11-24 21:02:56.552 node[3439:346410] EIO_Write
2016-11-24 21:02:56.552 node[3439:346410] EIO_Write1
2016-11-24 21:02:56.552 node[3439:346410] EIO_Write2
2016-11-24 21:02:56.552 node[3439:346410] EIO_Write3
2016-11-24 21:02:56.553 node[3439:346410] EIO_Write4
2016-11-24 21:02:56.553 node[3439:346410] getInstance..
2016-11-24 21:02:56.553 node[3439:346410] get 1
2016-11-24 21:02:56.553 node[3439:346410] get 2
2016-11-24 21:02:56.553 node[3439:346410] 20:16:10:19:11:45
2016-11-24 21:02:56.553 node[3439:346410] get 3
2016-11-24 21:02:56.553 node[3439:346410] get 4
2016-11-24 21:02:56.553 node[3439:346410] EIO_Write5
2016-11-24 21:02:56.553 node[3439:346410] to write...
write callback...
write...
2016-11-24 21:02:56.602 node[3439:346410] EIO_Write
2016-11-24 21:02:56.602 node[3439:346410] EIO_Write1
2016-11-24 21:02:56.602 node[3439:346410] EIO_Write2
2016-11-24 21:02:56.602 node[3439:346410] EIO_Write3
2016-11-24 21:02:56.602 node[3439:346410] EIO_Write4
2016-11-24 21:02:56.602 node[3439:346410] getInstance..
2016-11-24 21:02:56.602 node[3439:346410] get 1
2016-11-24 21:02:56.602 node[3439:346410] get 2
2016-11-24 21:02:56.602 node[3439:346410] 20:16:10:19:11:45
2016-11-24 21:02:56.602 node[3439:346410] get 3
// It stop here, by the way, I cannot tell if it is a crash or a hang issue. It just exit silently after this (no error log at all). In my test code, I can reproduce it about 1 out of 10 times to run. Different machine may get different results, it seems like an threading issue, but I have no idea why it stop at objectForKey function.

Appreciate if you have any idea how I can debug next step.

Thanks,
Shenghong

eelcocramer, 2016-11-25:
You can try debugging with [lldb](http://lldb.llvm.org) to see if it actually crashes there or there is a normal exit of the program. You can use this by:

```
lldb -- <path-to-node-executable> <path-to-your-script>
``` 

For example:

```
lldb -- /usr/local/opt/nvm/versions/node/v4.4.7/bin/node examples/async.js
```

You can start by issuing the `r` command in the debugger and just run the program from within the debugger.

gengshenghong, 2016-11-28:
@eelcocramer 
Thank you for your suggestions. I've tried lldb and also have some further progress now.

First of all, the issue is a seg fault 11 issue! (why I cannot see the seg fault previously? In previous code, the bluetooth communication is in a child process and it seems like the seg fault will not print in console. I was trying to cut down the case and now I can see it is a seg fault 11, the code logic is same and crash in same place).

See below logs:
2016-11-28 00:43:26.359 node[1542:184532] get 1
2016-11-28 00:43:26.360 node[1542:184532] get 2
2016-11-28 00:43:26.360 node[1542:184532] 20-16-10-19-11-45
2016-11-28 00:43:26.360 node[1542:184532] get 3
Process 1542 stopped
* thread #7: tid = 0x2d0d4, 0x00007fff868b70dd libobjc.A.dylib`objc_msgSend + 29, stop reason = EXC_BAD_ACCESS (code=1, address=0x18)
    frame #0: 0x00007fff868b70dd libobjc.A.dylib`objc_msgSend + 29
libobjc.A.dylib`objc_msgSend:
->  0x7fff868b70dd <+29>: andl   0x18(%r11), %r10d
    0x7fff868b70e1 <+33>: shlq   $0x4, %r10
    0x7fff868b70e5 <+37>: addq   0x10(%r11), %r10
    0x7fff868b70e9 <+41>: cmpq   (%r10), %rsi
(lldb) bt
* thread #7: tid = 0x2d0d4, 0x00007fff868b70dd libobjc.A.dylib`objc_msgSend + 29, stop reason = EXC_BAD_ACCESS (code=1, address=0x18)
  * frame #0: 0x00007fff868b70dd libobjc.A.dylib`objc_msgSend + 29
    frame #1: 0x00007fff8b979a4b CoreFoundation`-[__NSDictionaryM objectForKey:] + 139
    frame #2: 0x0000000102b10eb9 BluetoothSerialPort.node`+[BluetoothWorker getInstance:](self=<unavailable>, _cmd=<unavailable>, address=0x0000000102c0f080) + 169 at BluetoothWorker.mm:95
    frame #3: 0x0000000102b1342e BluetoothSerialPort.node`BTSerialPortBinding::EIO_Write(req=<unavailable>) + 186 at BTSerialPortBinding.mm:109
    frame #4: 0x0000000100a5572a node`worker + 90
    frame #5: 0x0000000100a61934 node`uv__thread_start + 25
    frame #6: 0x00007fff8e0d72fc libsystem_pthread.dylib`_pthread_body + 131
    frame #7: 0x00007fff8e0d7279 libsystem_pthread.dylib`_pthread_start + 176
    frame #8: 0x00007fff8e0d54b1 libsystem_pthread.dylib`thread_start + 13
(lldb)

It crashed every time at this place. Any idea?

Also, if I do not reply back in Arduino side, it will not crash! It seem like an multiple threading issue between write and receive.

Let me know if you have idea to share, I am still struggling on cutting down the code. :(

Thanks,
Shenghong

eelcocramer, 2016-11-28:
Can you post details on the version of nodejs & npm you are using? I would also like to know if you are testing this on a hackintosh.

eelcocramer, 2016-11-28:
Can you run the debugger again with the environment variable 'NSZombieEnabled=YES'?

gengshenghong, 2016-11-28:
@eelcocramer 

Nodejs version: v6.9.1, npm version: 3.10.8, included in official nodejs 6.9.1 package.

Also, I've tested your library 1.2.6 and latest 2.1.0, both have the same issue.

With "env NSZombieEnabled=YES" in lldb, the crash log is same. I've tested this on both hackintosh and   mac mini pro, same behavior.

I'll post a minimal test case below, hope you can reproduce in your side.
```
/*
 // filename: mac_issue_crash.js
 Test of bluetooth serial port library on Mac OS
 npm install bluetooth-serial-port@1.2.6 --registry=https://registry.npm.taobao.org
 npm install bluetooth-serial-port --registry=https://registry.npm.taobao.org
 
 Slave: HC06, baud rate 57600
 
 Setup:
 npm install bluetooth-serial-port --registry=https://registry.npm.taobao.org
 
 Reproduce:
 1. Upload Arduino Code
 2. in Mac OS X, run: node mac_issue_crash.js
 if it will not crash for a while (like 3 seconds), ctrl+C and try again, and again, and again...
 */
var btSerial = new (require('bluetooth-serial-port')).BluetoothSerialPort();

var address = "20-16-10-19-13-56";

var count = 0;
function doWrite() {
    // sysex_start cmd pin value sysex_end
    btSerial.write(new Buffer([240,8,2,false,247]), function(err, bytesWritten) {
                   if (err) {
                   console.log("Write error!");
                   console.log(err);
                   }
                   count = count + 1;
                   console.log(count);
                   setTimeout(function() {
                              doWrite(); // write next one
                              }, 20);   // really slow enough!
                   });
    
}

//btSerial.findSerialPortChannel(address, function(channel) {
channel = 1;
console.log("channel: " + channel);
btSerial.connect(address, channel, function() {
                 console.log('connected');
                 
                 // listen on the data
                 // btSerial.on('data', function(buffer) {
                 //  console.log(buffer);
                 //         });
                 
                 setTimeout(doWrite, 100);
                 }, function (err) {
                 console.log(err);
                 console.log('cannot connect');
                 });
```

The Arduino code:
```

#include <Firmata.h>

#define FM_OK               0x55

// commands
#define FM_CMD              0x08

// sub commands
#define FM_DIGITAL_WRITE    0x01

// return ok back
void sendOk(byte command) {
    Firmata.write(START_SYSEX);
    Firmata.write(FM_CMD);
    Firmata.write(command);
    Firmata.write(FM_CMD);
    Firmata.write(END_SYSEX);
}

void sysexCallback(byte command, byte argc, byte *argv)
{
  byte flag = 0;
  
  if (command == FM_CMD) {
        byte pin = argv[0];
        byte value = argv[1];
        pinMode(pin, OUTPUT);
        digitalWrite(pin, value);
        sendOk(command);
    }
}

void setup()
{
  Firmata.attach(START_SYSEX, sysexCallback);
  Firmata.begin(57600);
}

void loop()
{
 while (Firmata.available()) {
    Firmata.processInput();
  }
}
```

Without the "sendOk" in Arduino, it seems not reproducible (at least, not easy...I've tested many times).

How to reproduce:
```
TestdeMac:mac_issue_crash test$ node mac_issue_crash.js 
channel: 1
connected
1
2
...
49
50
51
52
^C
TestdeMac:mac_issue_crash test$ node mac_issue_crash.js 
channel: 1
connected
1
2
3
4
5
6
7
8
9
Segmentation fault: 11
TestdeMac:mac_issue_crash test$
```
(Keep run again and again, sometimes it is very easy to reproduce, sometimes, you may try 8 to 10 times.)

Let me know if you can reproduce and any quick fix?

Thanks,
Shenghong

gengshenghong, 2016-11-28:
By the way, when it crash, it usually stop at about 7 or 8 or 9 on my machine. Very stable...:(

Thanks,
Shenghong

gengshenghong, 2016-11-28:
FYI. In Arduino side, if I added "delay(100)" to delay 100ms before sendOK(command), it is difficult to reproduce (need more times to run...), and when it reproduced, it usually stop at about 28 or 29.

Thanks,
Shenghong

eelcocramer, 2016-11-28:
Can you try adding the following line just after [this one](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/osx/BluetoothWorker.mm#L79)?

```
[instances retain];
```

gengshenghong, 2016-11-29:
With this line, it still get seg fault 11. I've rebuilt with node-gyp.

Thanks,
Shenghong

gengshenghong, 2016-11-29:
If I added below:

```
BluetoothWorker *worker = nil;
    @synchronized (instances) {
       instances = nil;   // force it to alloc worker, it seems will not crash
        worker = [instances objectForKey:address];
        if (worker == nil) {
            worker = [[BluetoothWorker alloc] init];
            [instances setObject: worker forKey: address];
        }
    }
```

With this code, it will not crash (tested at least 30 times). But this code logic is incorrect, and data will not be send correctly. Also, as I mentioned above, if I do not reply back data in Arduino, it will not crash too. It seems like to be an issue of send and receive, they use same worker and maybe released somewhere by another (like in receive event?) It's just my guess based on the behaviour, as I had zero experience on obj-c and cannot understand some details in the code yet.

Thanks,
Shenghong

gengshenghong, 2016-11-29:
@eelcocramer More information, added some logs in EIO read/afterread, write/afterwrite, and the crash log as below (sometimes after EIO_Write, sometimes after EIO_Afterread). If I added "return" in EIO_Read, it will not crash. Hope it is helpful to get closer to the truth! It seems like send and receive thread conflicts and make the crash?

```
2016-11-28 19:26:36.983 node[1739:40132] EIO_AfterRead...
2016-11-28 19:26:36.983 node[1739:40141] EIO_Read...
2016-11-28 19:26:36.995 node[1739:40138] EIO_Write...
2016-11-28 19:26:36.996 node[1739:40132] EIO_AfterWrite...
7
2016-11-28 19:26:36.999 node[1739:40132] EIO_AfterRead...
2016-11-28 19:26:36.999 node[1739:40139] EIO_Read...
2016-11-28 19:26:37.021 node[1739:40140] EIO_Write...
2016-11-28 19:26:37.022 node[1739:40132] EIO_AfterWrite...
8
2016-11-28 19:26:37.026 node[1739:40132] EIO_AfterRead...
2016-11-28 19:26:37.026 node[1739:40138] EIO_Read...
2016-11-28 19:26:37.043 node[1739:40138] EIO_Write...
2016-11-28 19:26:37.044 node[1739:40132] EIO_AfterRead...
Segmentation fault: 11
TestdeMac:mac_issue_crash test$

6
2016-11-28 19:26:14.819 node[1734:38984] EIO_Write...
2016-11-28 19:26:14.819 node[1734:38975] EIO_AfterRead...
2016-11-28 19:26:14.819 node[1734:38983] EIO_Read...
2016-11-28 19:26:14.819 node[1734:38975] EIO_AfterWrite...
7
2016-11-28 19:26:14.845 node[1734:38982] EIO_Write...
2016-11-28 19:26:14.845 node[1734:38975] EIO_AfterWrite...
8
2016-11-28 19:26:14.849 node[1734:38975] EIO_AfterRead...
2016-11-28 19:26:14.849 node[1734:38981] EIO_Read...
2016-11-28 19:26:14.871 node[1734:38984] EIO_Write...
Segmentation fault: 11
TestdeMac:mac_issue_crash test$
```

eelcocramer, 2016-11-29:
I have the impression that somewhere in the read functions a buffer is overflowing which can be pretty hard to debug.

Can you try running your code with nodejs 4? You also need to recompile the module. Its just to rule out the possibility of a regression.

gengshenghong, 2016-11-29:
I cannot build with v4.4.7, below error:
ld: library not found for -lgcc_s.10.5.

Any idea?

Thanks,
Shenghong

gengshenghong, 2016-11-29:
@eelcocramer According to your information that there may be an buffer overflow in read. I've got below results:

```
// use "return" to stop the code logic to isolate the issue.
// works mean: put the "return" statement there, no crash, tested 30 to 40 times at least
// fail mean: put the "return" statement there, at least one crash reproduced.
void BTSerialPortBinding::EIO_Read(uv_work_t *req) {
    
    unsigned char buf[1024] = { 0 };

    NSLog(@"EIO_Read...");
    //return;
    
    read_baton_t *baton = static_cast<read_baton_t *>(req->data);
    size_t size = 0;
//return;  // works here
    memset(buf, 0, sizeof(buf));
//return;  // works here
    if (baton->rfcomm->consumer != NULL) {
        size = pipe_pop_eager(baton->rfcomm->consumer, buf, sizeof(buf));
    }
return;  // fail here
    if (size == 0) {
        pipe_consumer_free(baton->rfcomm->consumer);
        baton->rfcomm->consumer = NULL;
    }

    // when no data is read from rfcomm the connection has been closed.
    baton->size = size;
    memcpy(&baton->result, buf, size);
}

void BTSerialPortBinding::EIO_AfterRead(uv_work_t *req) {
    return;
    Nan::HandleScope scope;
NSLog(@"EIO_AfterRead...");
```

From above tests, the crash seems be caused by below lines:
```
 if (baton->rfcomm->consumer != NULL) {
        size = pipe_pop_eager(baton->rfcomm->consumer, buf, sizeof(buf));
    }
```

eelcocramer, 2016-11-29:
Looks similar to #76

eelcocramer, 2016-11-29:
Lets check if a newer version of the pipe solves this. There was an update of the code recently. I setup a branch that has this new version. Can you check if this changes anything for you?

https://github.com/eelcocramer/node-bluetooth-serial-port/tree/issue-125

eelcocramer, 2016-11-30:
@gengshenghong did you have a chance to check if the code in the `issue-125` branch fixes this issue?

gengshenghong, 2016-11-30:
@eelcocramer 
Thank you for your quick changes! It seems fix the issue exactly!!! I cannot reproduce the seg fault any more. :) I've tested many times and it is very stable now. Appreciate your work, please close this thread and looking forward for the release. 

Thanks,
Shenghong

eelcocramer, 2016-11-30:
Great, I will try to push a new version later today.

Thank you for letting me know!

eelcocramer, 2016-11-30:
This issue was fixed upstream in: cgaebel/pipe#5

eelcocramer, 2016-11-30:
There are some delays in the travis builds that are required so succeed before merge. The release has to wait until these are done. Hopefully the release can be done tomorrow morning (CET).

eelcocramer, 2016-12-01:
Merged and published to npm.

PR #126 (2016-12-01): 2.1.1 release
Fixes #125
closed 2016-12-01 by eelcocramer