eelcocramer/node-bluetooth-serial-port #127: Server: Bluetooth not visible

thatsgreat, 2016-12-01:
I am using the example server code from the Readme.

```
var server = new(require('bluetooth-serial-port')).BluetoothSerialPortServer();
var PORT = 10; // My service channel. Defaults to 1 if omitted.
var UUID = '38e851bc-7144-44b4-9cd8-80549c6f2912'; // My own service UUID. Defaults to '1101' if omitted

server.listen(function (clientAddress) {
    console.log('Client: ' + clientAddress + ' connected!');
    server.on('data' function(buffer) {
        console.log('Received data from client: ' + buffer);

        // ...

        console.log('Sending data to the client');
        server.write(new Buffer('...'), function (err, bytesWritten) {
            if (error) {
                console.log('Error!');
            } else {
                console.log('Send ' + bytesWritten + ' to the client!');
            }
        });
    });
}, function(error){
    console.error("Something wrong happened!:" + error);
}, {uuid: UUID, channel: PORT} );
console.log("listen");
```

It prints out "listen" and I expect the Bluetooth to be visible on my PC or Android phone. But I find nothing.

I am running the code on a Raspberry PI 3 using Raspbian and followed all steps of the setup in the Readme.

eelcocramer, 2016-12-01:
Can you set `PORT=1` and `UUID=1101` and retry?

thatsgreat, 2016-12-01:
Did so. Seems Bluetooth is still not discoverable.

eelcocramer, 2016-12-01:
@atilag can you help out here?

Juanjofp, 2017-02-02:
I do not know if it will be the same, but I had the bluetooth turned off, just turning it on worked

# bluetoothctl
[bluetooth] power on
[bluetooth] discoverable on

unlucio, 2017-05-15:
👍 
I'm having the same problem, with the example code.

BT is on and discoverable:

```sh
➜  btTest hciconfig -a
hci0:	Type: BR/EDR  Bus: UART
	BD Address: B8:27:EB:99:DF:DB  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN
	RX bytes:7651 acl:68 sco:0 events:197 errors:0
	TX bytes:4807 acl:66 sco:0 commands:129 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
	Link policy: RSWITCH SNIFF
	Link mode: SLAVE ACCEPT
	Name: 'raspberrypi'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous,
	HCI Version: 4.1 (0x7)  Revision: 0xb6
	LMP Version: 4.1 (0x7)  Subversion: 0x2209
	Manufacturer: Broadcom Corporation (15)
```

```sh
➜  btTest sudo service bluetooth status
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
   Active: active (running) since Mon 2017-05-15 19:18:47 UTC; 10min ago
     Docs: man:bluetoothd(8)
 Main PID: 1135 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─1135 /usr/lib/bluetooth/bluetoothd --compat --noplugin=sap

May 15 19:18:47 raspberrypi bluetoothd[1135]: Bluetooth daemon 5.23
May 15 19:18:47 raspberrypi bluetoothd[1135]: Starting SDP server
May 15 19:18:47 raspberrypi bluetoothd[1135]: Excluding (cli) sap
May 15 19:18:47 raspberrypi bluetoothd[1135]: Bluetooth management interface 1.14 initialized
May 15 19:18:47 raspberrypi systemd[1]: Started Bluetooth service.
```
The client part works fine, but when I launch the server it either just sits there or starts compaining about the SDP Server:

```sh
Something wrong happened!: Error: Cannot connect to SDP Daemon. errno: 13
```

adding a `setInterval()` to check on `server.isOpen()`  results in `false` 

here's an example server's output to sum it up:

```sh
➜  btTest node server.js
Server:  starting
BluetoothSerialPortServer status:  false
Something wrong happened!: Error: Cannot connect to SDP Daemon. errno: 13
BluetoothSerialPortServer status:  false
BluetoothSerialPortServer status:  false
BluetoothSerialPortServer status:  false
```

So, yeah, I'm kind of stuck as well.
Any hint is greatly appreciated, as well as I'm more than happy to carry on further debug is needed.

eelcocramer, 2017-05-15:
Wondering if this should run under `root` permissions.

unlucio, 2017-05-15:
It does solve the sdp problem:

```sh
➜  btTest sudo node server.js
Server:  starting
BluetoothSerialPortServer status:  false
BluetoothSerialPortServer status:  false
BluetoothSerialPortServer status:  false
```
```sh
root@raspberrypi:/home/pi/code/btTest# node server.js
Server:  starting
BluetoothSerialPortServer status:  false
BluetoothSerialPortServer status:  false
BluetoothSerialPortServer status:  false
BluetoothSerialPortServer status:  false
BluetoothSerialPortServer status:  false
```
but still nothing happens :(

atilag, 2017-05-15:
@unlucio So seems like server.isOpen() is not working properly. It returns "false" until some client actually connects. But yeah, this was not the intention, it should return "true" right after the server is listening... I'll fix it ASAP. In the meantime, check with another bluetooth capable device if the server is being announced... I see that your setup is good, so it should be working.

unlucio, 2017-05-15:
👍 now it shows up, but still no luck :(

[image]

eelcocramer, 2017-05-16:
Pairing != connecting to the server. After you paired you should try to connect with the example client.

scolestock, 2017-07-18:
For what it's worth: I've been able to get various Android bluetooth terminal utilities to connect to the "rfcomm-server.py" sample that comes with python-bluez (running on a raspi).  However, I can't get those same Android utilities to see the sample server (in this project's readme) with node-bluetooth-serial-port on a raspi.

vortex41, 2017-07-21:
It may be privileges issue. Try running
`# chmod 777 /var/run/sdp`
Remember also about running bluetoothd in compat mode - edit /etc/systemd/system/dbus-org.bluez.service and add --compat parameter in ExecStart:
`ExecStart=/usr/lib/bluetooth/bluetoothd --compat`

More info about probem here:
[https://raspberrypi.stackexchange.com/questions/41776/failed-to-connect-to-sdp-server-on-ffffff000000-no-such-file-or-directory](url)
and here:
[https://bbs.archlinux.org/viewtopic.php?id=201672](url)

closed 2018-04-12 by eelcocramer