eelcocramer/node-bluetooth-serial-port #128: Reported addres under Windows 10 is not only address

tzvetelin-vassilev, 2016-12-06:
Under Windows 10 reported address looks wrong. Under Windows 7, Windows 8 and Windows 8.1 field address contains exact address.

Inquiry results:

name: Tzvetelin�s iMac 
address: (8C:2D:AA:43:CD:E0)Tzvetelin�s iMac

name: Dragomir�s iMac
address: (10:40:F3:E4:C6:BA)Dragomir�s iMac

name: Wacom Sofia QA Dev SM
address: (20:D3:90:B7:C2:FF)Wacom Sofia QA Dev SM

List paired devices result:

{"name":"Tzvetelin�s iMac","address":"(8C:2D:AA:43:CD:E0)","services":[{"channel":1,"name":"SPP"}]}
{"name":"Wacom Sofia QA Dev (SM","address":"(20:D3:90:B7:C2:FF)","services":[{"channel":1,"name":"SPP"}]}
{"name":"Dragomir�s iMac","address":"(10:40:F3:E4:C6:BA)","services":[{"channel":1,"name":"SPP"}]}

tzvetelin-vassilev, 2016-12-06:
I have and question. In this code snippet:

    BluetoothSerialPort.prototype.close = function () {
        if (this.connection) {
            this.connection.close(this.address);
            this.connection = undefined;
        }

        this.emit('closed');
    };

this.connection.close(this.address) somehow breaks the app. it happens rarely but I cannot prevent this. I've tried to put try catch here but it was not successful. Also similar crash I'm receiving when inquiry happens. Also rarely. This methods calls binary. Is this could be handled somehow?

eelcocramer, 2016-12-06:
Hi, thank you for your report.

Are these 2 separate issues? If so please create a separate issue for each issue to keep the discussion clean and on topic. If not can you be more clear on how your comment relates to the issue.

As for your issue on Windows 10. What does the output you put in the issue show? Was it created on Windows 10 or another release? How is it different from another windows version?

I hardly ever use windows but Bluetooth addresses can be formatted differently on different platforms. An address looks different on Linux than on OSX as these use different byte separators.

eelcocramer, 2016-12-06:
cc: @langholz

eelcocramer, 2016-12-06:
As for you other comment. If this is another issue, please create one and give some more details about:

* What it is that make you think the program crashes on that line
* When you have it available, please include a crash log
* Details on the version of the module you are using together with the version of your operating system and the version of nodejs you are using.

Thanks!

tzvetelin-vassilev, 2016-12-07:
Hi,

Inquiry results comes from "found" event where address under windows 10 is something like this: (8C:2D:AA:43:CD:E0)Tzvetelin�s iMac
until in previous versions of windows address is: 8C:2D:AA:43:CD:E0

List paired devices comes from listPairedDevices method and address under windows 10 looks like this: (8C:2D:AA:43:CD:E0)
until in previous versions of windows address is: 8C:2D:AA:43:CD:E0

I understand delimiters are different under different operating systems, but result from windows 10 looks like some description, not address only.

For other question I'll move under another issue.

eelcocramer, 2016-12-07:
Ok, thank you for making this clear to me.

eelcocramer, 2016-12-07:
This seems like a bug to me. In [this line](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/windows/DeviceINQ.cc#L172) a discovered Bluetooth socket is converted to a human readable address string. The [following statement](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/windows/DeviceINQ.cc#L177) checks if this conversion is successful and if so [this line](https://github.com/eelcocramer/node-bluetooth-serial-port/blob/master/src/windows/DeviceINQ.cc#L180) tries to get the bluetooth address from the human readable string. If that last line fails the human readable string is used and this seems to happen on your Windows 10 setup.

I'm not sure why this is happening and I'm also not able to fix this myself as I'm not familiar with Windows programming and more important I do not have a Windows development environment available.

Hopefully some one else is able to look into this further.