eelcocramer/node-bluetooth-serial-port #130: A dynamic link library (DLL) initialization routine failed.

Nols1000, 2016-12-12:
```
Uncaught Error: A dynamic link library (DLL) initialization routine failed.
\\?\C:\Users\*\node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPort.node
    at Object.Module._extensions..node (module.js:640:18)
    at Module.load (module.js:512:32)
    at tryModuleLoad (module.js:471:12)
    at Function.Module._load (module.js:463:3)
    at Module.require (module.js:522:17)
    at require (internal/module.js:20:19)
    at bindings (C:\Users\*\node_modules\bluetooth-serial-port\node_modules\bindings\bindings.js:76:44)
    at C:\Users\*\node_modules\bluetooth-serial-port\lib\bluetooth-serial-port.js:20:39
    at Object.<anonymous> (C:\Users\*\node_modules\bluetooth-serial-port\lib\bluetooth-serial-port.js:135:2)
    at Module._compile (module.js:595:32)
```

I installed the module on Windows 10 and using it with nwjs.io.

eelcocramer, 2016-12-13:
Does that file exist on your system? I.e. `\\?\C:\Users\*\node_modules\bluetooth-serial-port\build\Release\BluetoothSerialPort.node`

Nols1000, 2016-12-13:
Yes. Also turned it off and on again ;D

eelcocramer, 2016-12-13:
Just checking :-P

Did you install using these instructions?

http://docs.nwjs.io/en/latest/For%20Users/Advanced/Use%20Native%20Node%20Modules/

If I google your error I get some posts about not having nw-gyp setup correctly.

eelcocramer, 2016-12-15:
Did you get it to work?

Nols1000, 2016-12-16:
Yes.

closed 2016-12-16 by Nols1000

eelcocramer, 2016-12-16:
Nice. Thank you for the feed back.