eelcocramer/node-bluetooth-serial-port #131: closed event fired an incrementing number of times

andreavaresio, 2017-01-03:
Every time a client disconnects, closed event is fired once more.
The first time once, the second time twice and so on. 
Here the code I used. I used the dirty trick to reschedule the server.listen with a delay after client disconnection.

`var server = new(require('bluetooth-serial-port')).BluetoothSerialPortServer();
var CHANNEL = 1; // My service channel. Defaults to 1 if omitted.
//var UUID = '38e851bc-7144-44b4-9cd8-80549c6f2912'; // My own service UUID. Defaults to '1101' if omitted
var UUID = '1101';
var gEnableServerReload = true;

function PollServerStatus()
{
	if (gEnableServerReload)
		HandleAConnection();
	setTimeout(PollServerStatus, 500);
}

function HandleAConnection()
{
	gEnableServerReload = false;
	console.log('Waiting for connection...');
	server.listen(function (clientAddress) {
		console.log('Client: ' + clientAddress + ' connected!');
		server.on('data', function(buffer) {
			console.log('Received data from client: ' + buffer);

			// ...

			console.log('Sending data to the client');
			server.write(new Buffer('ciao'), function (err, bytesWritten) {
				if (err) {
					console.log('Error!');
				} else {
					console.log('Send ' + bytesWritten + ' to the client!');
				}
			});
		});
	}, function(error){
		console.log("Something wrong happened!:" + error);
	}, {uuid: UUID, channel: CHANNEL} );
	server.on('closed', function(){
		console.log('Client: disconnected!');
		gEnableServerReload = true;
	});
}
PollServerStatus();
`

eelcocramer, 2017-01-04:
Is this the same issue as #115?

eelcocramer, 2017-01-04:
You are adding new event listeners to `server` every time you call `HandleAConnection()`. Move `server.on('closed')` up to the `PollServerStatus()` function and your problem is propably fixed.

andreavaresio, 2017-01-04:
Thanks a lot - problem solved

eelcocramer, 2017-01-04:
Cool.

closed 2017-01-04 by eelcocramer