eelcocramer/node-bluetooth-serial-port #136: program crashes (Windows/Linux) if no Bluetooth available (no Bluetooth Hardware/Bluetooth deactivated)

danneboom, 2017-01-24:
On Windows and Linux (raspberry pi 2) the package crashes if there is no bluetooth stick connected (raspberry pi) or deactivated (windows).
I use the following code:
```
btSerial = new (require('bluetooth-serial-port')).BluetoothSerialPort();
btSerial.inquire();
        btSerial.on('finished', function (){
            setTimeout(function () {
                btSerial.inquire(); //search again
            },10000); //bluetooth search interval
        });
```
the program crashes at btSerial.on(..). Is there any workaround?
(If i activate BT / use BT Stick everything works fine).
Thanks!

eelcocramer, 2017-01-24:
Is there a stack trace?

As a workaround in Linux [you can check on forehand if your device has bluetooth](http://askubuntu.com/questions/301458/how-can-i-tell-if-my-laptop-has-bluetooth). You can probably call this from a nodejs script as well.

danneboom, 2017-01-25:
No, the process just finishes, even with try-catch-statement around. There is no error thrown.
Thanks for the workaround.

gengshenghong, 2017-04-07:
@eelcocramer The crash is caused by below code:
bt_inquiry DeviceINQ::doInquire in Linux/DeviceINQ.cc for example:
`dev_id = hci_get_route(NULL);
  sock = hci_open_dev( dev_id );
  if (dev_id < 0 || sock < 0) {
    Nan::ThrowError("opening socket");
  }`

I can reproduce the error on Linux Ubuntu.

the "doInquire" in called in Execute in InquireWorker, which cannot catch the throwerror (I've searched in Google, and it was mentioned that you need to use SetErrorMessage and HandleErrorCallback to handle the errors in async worker). Hope these information are helpful for you.

When using inquireSync, we can use try/catch to catch the throwerror in user code, but using inquire, it will crash internally (seg fault).

Thanks,
Shenghong

jdpigeon, 2018-04-19:
Found this on a Windows machine. Gonna put some try catch statements around `btSerial.inquire()` in order to prevent it from crashing the whole app

closed 2020-02-06 by eelcocramer